pimusicbox/pimusicbox #395: New IQaudio and HiFiBerry and Pimoroni boards?

shawaj, 2017-03-16:
Might be worth getting some input from @iqaudio and @hifiberry and @pimoroni here....as there a number of boards that could do with being supported in a new release:

- [IQaudio Pi-Digi+](http://iqaudio.co.uk/audio/48-pi-digi.html)
- [HiFiBerry DAC+ Pro](https://www.hifiberry.com/shop/boards/hifiberry-dac-pro/)
- [HiFiBerry Digi+ Pro](https://www.hifiberry.com/shop/boards/hifiberry-digi-pro/)
- [Pimoroni Speaker pHAT](https://shop.pimoroni.com/products/speaker-phat)
- [Pimoroni pHAT BEAT](https://shop.pimoroni.com/products/phat-beat)

There are also new Zero boards from @iqaudio and @hifiberry but I am guessing these are already supported by their existing drivers for DAC/Amp boards?

shawaj, 2017-03-16:
Possibly even some other boards from Allo, Audiophonics and a few others too that might warrant inclusion?

shawaj, 2017-03-28:
Spoke to the people at Audiophonics.

These DACS:

- I-Sabre V2 : http://www.audiophonics.fr/fr/dac-diy/audiophonics-i-sabre-dac-es9023-v2-tcxo-raspberry-pi-20-a-b-i2s-p-10176.html
- I-Sabre V3 : http://www.audiophonics.fr/fr/dac-diy/audiophonics-i-sabre-dac-es9023-v3-tcxo-raspberry-pi-20-a-b-i2s-p-10657.html
- I-Sabre AMP : http://www.audiophonics.fr/fr/dac-diy/audiophonics-i-sabre-amp-dac-es9023-amplificateur-class-d-2x30w-tpa3118-p-11800.html
- I-TDA 1387 : http://www.audiophonics.fr/fr/dac-diy/audiophonics-i-tda1387-tcxo-dac-raspberry-pi-3-a-b-pi-2-i2s-p-11585.html

Are all compatible with either the "HiFiBerry DAC" or "RPI DAC" drivers.

shawaj, 2017-03-28:
Speaker pHAT and pHAT BEAT appears to use "hifiberry-dac" and "i2s-mmap" dtoverlays in /boot/config.txt so it appears that it just needs that hifiberry-dac driver - same as the pHAT DAC.

@pimoroni @gadgetoid - is that correct?

@kingosticks is it just settings.ini and setsound.sh that would need to be changed in this instance? and mopidy_websettings/index.html over at https://github.com/pimusicbox/mopidy-websettings/ ?

shawaj, 2017-03-28:
According to this page on HiFiBerry website - https://support.hifiberry.com/hc/en-us/articles/205377651-Configuring-Linux-4-x-or-higher

Also see here - https://github.com/raspberrypi/linux/blob/1e628b3e69823119ed5441ca4bfe9e32bd63d6f9/arch/arm/boot/dts/overlays/README

It seems that DAC+ Pro uses hifiberry-dacplus overlay
Digi+ Pro uses hifiberry-digi-pro overlay although may still work with hifiberry-digi overlay?

@hifiberry @bonezuk is this correct? don't have them to test

shawaj, 2017-03-28:
IQAudio Pi-Digi+ seems to use overlay iqaudio-digi-wm8804-audio
IQAudio Pi-DAC Pro seems to use existing overlay iqaudio-dacplus
IQAudio Pi-DAC Zero seems to use existing overlay iqaudio-dacplus

@iqaudio can you confirm?

kingosticks, 2017-03-28:
@shawaj Yes. Although with all the products using the same driver I'm inclined to not bother adding them. Anyone who buys a pHAT DAC will probably know to select a hifiberry-dac, for example. Is it worth it?

shawaj, 2017-03-28:
@kingosticks Yeah, I see your point. Perhaps easier to just add to an FAQ
then, if not wanting to support them all with their names?

So then I guess it's just support for the new overlays that aren't
currently supported that needs including?

And maybe getting some people with access to the mentioned boards to test
them out...

kingosticks, 2017-03-28:
Yes new overlays only I think. Although PRs would always be welcome. I'll add something to the websettings text to that effect. And also add it to my huge list of doc updates to do. 

Testing them would be nice. 

I was intending to have a mechanism to allow websettings to be more data driven with a config that controlled the settings page. So then all the updates for this kind of thing are limited to pimusicbox, It's a bit annoying changing it in both places.

shawaj, 2017-03-30:
Happy to do the PR. What files would need changing?

And also happy to help with doc updates. Is this list public somewhere?

kingosticks, 2017-04-01:
Exactly the ones you correctly identified before:
> is it just settings.ini and setsound.sh that would need to be changed in this instance? and mopidy_websettings/index.html over at https://github.com/pimusicbox/mopidy-websettings/ ?

A list is a great idea. See #405

kingosticks, 2017-04-01:
Having said that, I just had to look up what to select for an Amp Zero. That's hardly in the spirit of Pi Musicbox, maybe we can add cards like that just to websettings, as long as the `<select>` value is right then setsound.sh doesn't need to know anything. That probably makes the most sense.

iqaudio, 2017-04-02:
Hi Nick, wrt IQaudIO boards.

There are two board drivers, the pcm512x DAC+ driver and the Wm8804 digi+ driver.

The DAC+ driver can be used across Pi-DAC+, Pi-DACZero, Pi-DAC PRO and Pi-DigiAMP+. It should be used with the unmute parameter to ensure the amp stage is operational for immediate music playback.
dtoverlay=iqaudio-dacplus,unmute_amp 

The Digi+ driver is for our version of the wm8804 reference design. The Pi-Digi+
dtoverlay=iqaudio-digi-wm8804-audio 

Overlay configs (if needed) are documented on page 3 of our user guide:
www.iqaudio.com/downloads/IQaudIO.pdf

If you have any questions then please do PM me.

Thanks in advance,

Gordon

shawaj, 2017-04-02:
@iqaudio just to check having the unmute_amp parameter is ok for the DAC
boards as well?

>
> Having said that, I just had to look up what to select for an Amp Zero.
That's hardly in the spirit of Pi Musicbox, maybe we can add cards like
that just to websettings, as long as the <select> value is right then
setsound.sh doesn't need to know anything. That probably makes the most
sense.

iqaudio, 2017-04-02:
yes, it’s used for Pi-AMP+ unmute. No issue.

shawaj, 2017-04-04:
FYI....Allo DACs....

Name:   allo-boss-dac-pcm512x-audio
Info:   Configures the Allo Boss DAC audio cards.

Name:   allo-piano-dac-pcm512x-audio
Info:   Configures the Allo Piano DAC (2.0/2.1) audio cards.
        (NB. This initial support is for 2.0 channel audio ONLY! ie. stereo.
        The subwoofer outputs on the Piano 2.1 are not currently supported!)

Name:   allo-piano-dac-plus-pcm512x-audio
Info:   Configures the Allo Piano DAC (2.1) audio cards.

@babuenir @raashidmuhammed - is this correct?

babuenir, 2017-04-04:
@shawaj - Yes, you are correct. The subwoofer outputs are supported in "allo-piano-dac-plus-pcm512x-audio"

whnr, 2017-05-15:
@shawaj I just tried to boot a RPi Zero W with @hifiberry DAC+ Zero
According to their [configuring linux](https://support.hifiberry.com/hc/en-us/articles/205377651-Configuring-Linux-4-x-or-higher) article you linked above the [Hifiberry Dac+ Zero](https://www.hifiberry.com/shop/boards/hifiberry-dac-zero/) needs to be set to `hifiberry-dac`

Somehow my complete system was fried afterwards… I'll let you know if it works.

tomhumphrey, 2017-05-24:
@wnhr I've just tried exactly the same and can't get any sound out. Nor can I get system settings on the musicbox to stick (e.g. setting default volume, turning on AirPlay, and setting audio card from automatic to any other setting)

kingosticks, 2017-05-24:
@hifiberry it could be a lot clearer which overlay is required for your DAC+ Zero as that article doesn't actually explicitly mention the DAC+ Zero anywhere.

It seems you should be setting `output = hifiberry_dac` in settings.ini or selecting "HifiBerry DAC / pHAT DAC" from the websettings drop down menu. This assumes you are using at least v0.7RC4. Save the file and reboot cleanly.

Setting the wrong value should not cause anything to be "fried" (and I'm confused how a complete system could recover from such a thing... perhaps we have different interpretations of that word)

tomhumphrey, 2017-05-24:
I've tried this both on the SD card direct and through the settings page and it keeps getting over written. (I installed the v0.7RC4) Either way I don't get any sound out of the board

tomhumphrey, 2017-05-24:
Tried with a new SD card. Set all the settings before first boot and it's working like a charm 👍

kingosticks, 2017-10-04:
Seems I missed the Allo DAC firmware files by a [few days](https://github.com/raspberrypi/firmware/commits/master/boot/overlays/allo-boss-dac-pcm512x-audio.dtbo) and those cards do not work in RC5. Need to update the firmware to something newer than April 3rd (kernel 4.9.20).  

Thanks to clriis for reporting [here](https://discuss.mopidy.com/t/boot-loop-enabling-dtoverlay-pi-dac/1816/14?).

ytzelf, 2018-02-07:
Has this been cleared since - I see current version is still RC5?
Thanks a lot

kingosticks, 2018-02-07:
I have not released RC6 yet.

davidblumntcgeo, 2018-03-24:
@shawaj @kingosticks @hifiberry Just out tried Musicbox 0.7RC6 (#473). I'm running a RPi 3B with Hifiberry Digi+ Pro card, and I configured the output device as Hifiberry Digi from the dropdown menu. Playback of tracks encoded at 44.1, 88.2, and 176.4 kHz works fine. Playback of tracks encoded at 48, 96, and 192 kHz was buggy. Is the hifiberry-digi-pro overlay available, or is there another identical overlay for a different I2S card that I can use? If not, could you kindly consider this as an additional feature in the next release? I have no issues with playback of tracks at 48, 96, and 192 kHz using Moode OS, so the problem is not with my hardware. Thank you!

commit 23026b6e (2017-04-19): Support additional overlay params. Fixes #395.