pimusicbox/pimusicbox #396: Can you put Pi Music box play songs of videos from Youtube?

sami10007, 2017-03-16:
[Here](https://www.youtube.com/watch?v=T_Dy9OvgjHY) an example of video for the input.

kingosticks, 2017-03-16:
Yes. It uses mopidy-youtube, which does exactly this.

sami10007, 2017-03-17:
How stable is the solution? - - I am thinking how many times has the approach broke because of updates in Youtube.

kingosticks, 2017-03-17:
It's been solid lately but it's an unofficial API and I can't predict what youtube will do.

kingosticks, 2017-03-18:
There's no actual issue here so closing. 

If you have any other questions about mopidy-youtube then please direct them at the project directly. 

If you have any other questions about using pimusicbox then please post them at https://discuss.mopidy.com/c/pi-musicbox

closed 2017-03-18 by kingosticks

dhetriteq, 2018-02-22:
Youtube is still not working