pimusicbox/pimusicbox #398: Shairport not configured in 0.7RC4 (maybe upnp as well)

Airforc, 2017-03-21:
@kingosticks ,

Downloaded RC4 today out of the box all backend services I have tried have worked on the web front end (Gmusic, Spotify). For Airplay it was not reporting a target. Checked config, default config not set. 

Solution: Uncomment the first line in general of /etc/shairport-sync.conf , reboot.

kingosticks, 2017-03-21:
Hi, thanks for the report. I did try airplay last night but I didn't see this issue. And the commented out values are just the default values so it's a bit weird. Are you sure it wasn't just the restart that made it work? How did you enable it? Through editing settings.ini or websettings?

kingosticks, 2017-03-21:
I just tried again with a fresh sd card and i am not seeing this.

prasquin, 2017-03-24:
Hi,
It's also doesn't work with me with a fresh install of RC4.
I'm using macOS Sierra 10.12.3 and IOS 10.2.1. They see musicbox but cannot connect to it.

With this tutorial: https://learn.pimoroni.com/tutorial/sandyj/streaming-airplay-to-your-pi
without pimuscibox everything works fine.
Between the two installation  /etc/shairport-sync.conf is exactely the same

kingosticks, 2017-03-24:
@prasquin Would you mind running the following commands to see if the firewall is getting in the way:
```
iptables -P INPUT ACCEPT
iptables -P OUTPUT ACCEPT
iptables -P FORWARD ACCEPT
iptables -F
iptables -X
```

prasquin, 2017-03-24:
ok now it's working fine

kingosticks, 2017-03-24:
I guess different versions of airplay require different ports. This is unfortunate.

Airforc, 2017-03-26:
Sorry for the delay,

I think this might be two separate issues. My concern is in the shairport-sync.conf a name is required to be setup something like name = "%H"; which will name it whatever the hostname is. by default on the RC4 there is nothing setup it is just a default config that shows the options that are available with everything commented out. This happened with the first use on my Pi2. I just tried on a PiZeroW and do not have this problem. Will continue investigating.

kingosticks, 2017-03-26:
I appreciate that but the value that's commented out is the default value that the option will use. I.e if you don't set anything it'll use the hostname. Nothing is required. I really suspect what you saw before was something else. But I have hijacked your original issue which I apologise for, I'll make a separate one for the firewall problem and rename this.

Airforc, 2017-03-26:
No problem, like I said. I experienced it with a Pi2 and a fresh image. My pi zero does not have the same issue. I think we might be able to close this issue.....I am going to try on my Pi2 and my Pi3 with fresh images and see if it happens again.

kingosticks, 2017-03-26:
Ok great, that'd be useful. Thanks

kingosticks, 2017-04-01:
Were you able to reproduce this?

Airforc, 2017-04-04:
@kingosticks , 

Have been busy at work. I will try to reproduce by the end of the week.

commit ff70f0e0 (2017-04-11): Ensure have network before starting services that require it. Fixes #398.
closed 2017-04-11 by kingosticks