pimusicbox/pimusicbox #399: Spotify playlists missing from v0.7.0RC4

jim-miller-gac, 2017-03-25:
Spotify playlists are not appearing for me in 0.7RC4. I can make a pimusicbox playlist using the new feature in the queue area. That shows up and plays fine. But no playlists coming in from Spotify.

I've also tried a fix that has worked for me in the past. Comment out the return in the session_manager.py script file:

    def refresh_playlists(self):
    """Refresh the playlists in the backend with data from Spotify"""
    if not self.initialdata_receive_completed:
    logger.debug('Still getting data; skipped refresh of playlists')
    #return
    playlists = []
    folders = []

But that fix has no effect here. Also tried restarting the Mopidy service. No luck...

Please advise. Thank you.

kingosticks, 2017-03-26:
That 'fix' is already present in the image. I did find that older (slower) Pis did require a wait before things were loaded and became usable. But they did always load in the end. Did you try leaving it for a bit before restarting?

jim-miller-gac, 2017-03-26:
Tried again today. From boot, waited 30 minutes. No change. I think I'll try this on a RPi2 and report back...

jim-miller-gac, 2017-03-28:
Continued my hunt for Spotify playlists with an RPi2 (0.7RC4). Tried both wireless and wired internet (without an Edimax wireless adapter). Tried my wife's side of our family Spotify account (thought something could be corrupted in my playlists; note I use playlist folders on Spotify). Even tried Volumio on the RPi2. In all cases, same problem, no playlists. Starting to daydream about the good old days of vinyl... Had a couple good years with my RPi and PiMusicBox, even wrote about it [here](http://pet.timetocode.org/raspberrypi.html). But starting to be drawn to an approach where Spotify is installed on the RPi. MIght try a VM like ExaGear Desktop.

Will try going back to 0.6 but on the RPi2...

kingosticks, 2017-03-28:
Spotify have stopped supporting their library and have not replaced it so we are stuck with their bug-infested mess that's seen no updates in 2+ years. They've promised a new library but delivered nothing. 

I've not seen this problem but I've had a report of large numbers of playlists causing issues. Are you able to try with another account that has less? 0.7 and 0.6 are very very similar in terms of Spotify support so I would be surprised if that helps. 

There are a couple of things you could try if you still have the interest but neither work under musicbox (yet). 

* install the very latest mopidy and mopidy-spotify on a fresh jessie install. The playlist loading is slightly different (but still relies on libspotify) and the debug logging is much better, might show the issue.
* try librespot (again debian jessie only), a reverse engineered implementation of Spotify connect. This must be controlled with an official client e.g. running on your phone. This does not use libspotify but may have its own set of bugs! 
* or, install spotify-connect-web. This *does* work on musicbox but is very buggy. I've had mixed success with it...

jim-miller-gac, 2017-03-28:
I have about 170 playlists. Some are very long ones like "The Rolling Stones in Mono." My wife's side of the account has only a handful of lists, that's why I also tested there. But maybe our marriage extends to Spotify and our separate individual accounts are really still connected under our family account. I'm reluctant to delete my lists to further test (they probably feed the aggregators and help define a signature for me). But I see Spotify has a feature to recover deleted playlist... Thank you for the suggestions. And thank you for your patience and kind responses through all this LibSpotify stuff.

kingosticks, 2017-03-28:
I thought family accounts had up to 4 members? If so, you could create a new account just to test. I wouldn't want to delete my playlists!

jim-miller-gac, 2017-03-30:
Have found this Volumio plugin for Spotify Connect to be very good (playlists are accessible via Spotify clients).

Import the volspotconnect2.zip file from this page:
https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2

anttiautio, 2017-03-31:
Hi, I built a Raspberry Pi 3 / Hifiberry DAC+ player a week ago. My Spotify playlists loaded fine initially, but today I noticed they had disappeared from MusicBox.

kingosticks, 2017-03-31:
Which client are you using?

anttiautio, 2017-03-31:
I'm not sure if I understood the question right (sorry, I'm totally new to this) but I have all the Spotify components enabled and they seem to work fine. Only the playlists have disappeared. I have not changed my configuration since I installed the audio player last weekend.

kingosticks, 2017-03-31:
I meant which of the web clients or mpd clients are you using (there are multiple to choose from). But i think that means you are using the default musicbox_webclient.

anttiautio, 2017-03-31:
Yes, I use the default web client, haven't tried the others yet. 

By the way thanks for your great work on Pi Music Box! It is a very elegant and easy-to-use package to build a simple RPi music player -- in other words exactly what I need.

kingosticks, 2017-03-31:
Does a reboot (or Mopidy restart) bring them back?

kingosticks, 2017-03-31:
Also, if you are having Spotify playlist problems and only really use Spotify, then try changing to the mopify web client. If I remember correctly it loads the playlists from Spotify directly itself so shouldn't suffer from this. Plus it also has some other extra Spotify-specific features.

anttiautio, 2017-03-31:
I restarted the MusicBox when I noticed that the playlists were gone, but that did not help. Thanks for the tip re: Mopify, I'll have to try that out at some point. I only use Spotify and local music files. 

Come to think of it, the only adjustments I have made after installing the box was copying some music files to the SD card and enabling the scan on boot feature. Plus I added a cron job to run shutdown -r now at 6 am so that new files get scanned while I'm sleeping (no idea if that is really necessary).

anttiautio, 2017-03-31:
Hi again, yes -- Mopify seems to load the playlists. Unfortunately I did not find an option to play local files, and I don't really like the UI, so I think I'll keep using the default webclient or Moped and live without the playlists for a while :) (Moped does not load the playlists either.)

kingosticks, 2017-03-31:
Yeh, only Mopify loads the playlist directly and it's very much focused on Spotify. Maybe https://github.com/jaedb/Iris does too (but that's not not installed in Musicbox... yet!).

rmlinnovator, 2017-03-31:
I too am having this problem with .70 RC 4.  No spotify playlists.  I tried a couple of other Mopidy web clients and they too don't show the playlists.  I tried restarting the Mopidy service.  I have an older pi. Raspberry PI Project Board Rev. B

kingosticks, 2017-04-01:
As I've tried to say, **only** mopify has the ability to load playlists directly from Spotify. All other clients load the playlists through Mopidy. So whatever this problem is, it'll be an issue for every single web client except mopify. I have been unable to reproduce it so far.

spotify-connect-web is not installed in 0.7.0RC4 so I don't understand what you are referring to

anttiautio, 2017-04-03:
Wow, I just noticed my playlists have mysteriously returned to the default MusicBox webclient! I have absolutely no idea what is going on. I don't remember changing any settings in the past few days, but the MusicBox has performed a couple of scheduled restarts (I use a cron job to restart it every morning to rescan local files).

anttiautio, 2017-04-04:
Today: playlists missing again.