pimusicbox/pimusicbox #400: Slow streams

RebelDesigner, 2017-03-25:
In the latest RC streams tend to start extremely slow, sometimes even a 30 second delay.

kingosticks, 2017-03-26:
I'm not sure what we can do about that, I would guess it's very dependent on what stream you are loading. If there are many nested playlists for Mopidy to unwrap then it does this differently (and more reliably) in the newer version included in 0.7.0RC4, which takes longer.

RebelDesigner, 2017-03-27:
Thank you for your reply. I have added the direct URIs for the stream (so no .m3u) and the same problem. To show what I mean: when I `curl` the streaming URI it starts streaming binary data in my (Mac) terminal console and doesn't return a playlist.

Is this also expected behaviour?

kingosticks, 2017-03-27:
No, certainly not. Can you give me the stream URI?

RebelDesigner, 2017-03-27:
For example http://streaming.radionomy.com/Toppers-van-Toen

kingosticks, 2017-04-01:
I tried this and it took approx 10 seconds from selecting play to hearing anything. Most of this time is spent buffering the station, during which the webclient decides to hammer Mopidy with `get_time_position` requests, which probably doesn't help matters. 

I then tried the exact same thing in mopidy-mobile (which doesn't go crazy with `get_time_position`) but still had to wait the same length of time, so it looks like this is purely a buffering problem and musicbox-webclient is off the hook. 

You can see in [my log](http://dpaste.com/1TQJX1J) all the expected loading happens quite quickly but then there's nothing happening for 8 seconds from 19:20:11,853 until 19:20:19,509 at which point it's 100% buffered and starts playing. 

```
< loads of reasonable looking stuff going on >
2017-04-01 19:20:11,851 DEBUG [2694:MainThread] mopidy.audio.gst: Got async-done.
2017-04-01 19:20:11,853 DEBUG [2694:MainThread] mopidy.listener: Sending state_changed to AudioListener: {'old_state': u'stopped', 'target_state': u'playing', 'new_state': u'paused'}
2017-04-01 19:20:19,509 DEBUG [2694:MainThread] mopidy.audio.gst: Got buffering message: percent=100%
```

So this is a Mopidy issue, either one of the many Gstreamer0.10 problems or possibly something related to https://github.com/mopidy/mopidy/issues/1280 (this stream also keeps randomly buffering just like that issue). Unfortunately since we are running Mopidy v1.x we first need to try this out in the latest Mopidy version (on regular Raspbian) to work out which it is.

If you can get a debug log I expect you would see something similar, but maybe it's just taking a further 20 seconds to buffer for you? If you're interested you can grab the improved debugging logging settings for your image:
```
wget -q -O /etc/mopidy/logging-debug.conf https://raw.githubusercontent.com/pimusicbox/pimusicbox/8975e173426b035d4f0964c0175c9e422563a64a/filechanges/etc/mopidy/logging-debug.conf
```
Then enable it by changing the logging config line in settings.ini to the following:
```
[logging]
config_file = /etc/mopidy/logging-debug.conf
```

I recommend also disabling the DLNA/UpNP service (upmpdcli) since that spams the log full of uninteresting status messages and makes it more confusing. You'll find the debug log in `/tmp/mopidy-debug.log`. I tend to revert settings.ini when I'm done but it's up to you.

kingosticks, 2017-04-04:
There's a similarly long wait when using Mopidy v2.1.0 of approx 9 seconds. It's the buffering again: http://dpaste.com/0TV8303

kingosticks, 2017-04-04:
And with the set_live(True) proposal (https://github.com/mopidy/mopidy/issues/1280) it starts playing in about 5 seconds. We could try that...

melvinkcx, 2018-07-22:
I am experiencing it when using it as a DLNA client.