pimusicbox/pimusicbox #402: Switching stream after some times breaks webclient

RebelDesigner, 2017-03-27:
In the latest RC there might be a problem with the webclient or something else.
After playing a stream for some time, and then switching to another stream, doesn't do anything. When reloading the webclient, Musicbox cannot be found ("Trying to reach Musicbox..."). Seems like something is crashing. After some time and a refresh, everything starts working again. Haven't had the opportunity to dig any further. Any logs I can post?
I can replicate the problem (most of the time) on two RPi's.

javieitez, 2017-04-03:
I've replicated the same problem as well.

kingosticks, 2017-04-03:
And we think this is a different issue to https://github.com/pimusicbox/pimusicbox/issues/400 ? Would one of you be able to try getting some extra logging info as described in that linked issue?

RebelDesigner, 2017-04-03:
It might be related, but I don't think it is exactly the same issue.
I'll try tonight to get the debug logs in.
@javieitez: could you maybe post your exact steps? I haven't done that; I don't run into this every time and there are no clear steps for me to reproduce this. Would also help to identify the issue better.

javieitez, 2017-04-06:
Well, all I can say is that after some time streaming from any of the Tune In stations, the http frontend stops loading. It still works like a charm from an Android client.

[image]

javieitez, 2017-04-06:
Hi,

After streaming for a while from any of the TuneIn stations, the http frontend returns _"Trying to reach Musicbox..."_ , but waiting one minute and refreshing the page seems to reload everything correctly.

I've tried monitoring the CPU load with htop while doing this, but it never goes over 85%.

kingosticks, 2017-04-06:
Does the audio stop during this period? I really need a log of this.

javieitez, 2017-04-06:
No, it doesn't stop playing audio at all. It just stops sending data to the http frontend on port 80, temporarily. It responds to an android MPD client, in port 6680

The log

```
2017-04-03 03:33:51,110 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2017-04-03 03:33:52,031 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2017-04-03 03:37:26,439 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2017-04-03 03:43:47,529 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
```

kingosticks, 2017-04-06:
Sorry, I mean a DEBUG log as described in https://github.com/pimusicbox/pimusicbox/issues/400#issuecomment-290943063 (it's a bit of a faff)

kingosticks, 2017-04-06:
My guess was that the stream starts buffering and then the webclient starts erroneously sending a load of requests for the current time position (see https://github.com/pimusicbox/mopidy-musicbox-webclient/issues/237) which create a backlog and stop it from responding. But then I would expect the audio to cut out. Unless perhaps it can fill the buffer back up fast enough. I don't to see any buffering problems on my connection at work which might explain why I've not noticed it.