pimusicbox/pimusicbox #405: Documentation updates required for v0.7.0

kingosticks, 2017-04-01:
The website and nearly all of the current documentation requires updating. I don't want to spend too much time on it since it's only likely to be relevant for the 0.7.x release(s). I've already made some changes to the website (https://github.com/pimusicbox/pimusicbox.github.io/commit/fdf60af8d5d8c77065d4a124140ed4cdabe7ba67) but the user guide needs similar treatment. As long as we fix the things that are noticeably wrong/outdated then that would probably be sufficient right now.

- [ ] Update installation guide (maybe just to match website)
- [ ] Remove/change upgrade page
- [ ] Make clear that streamuris has changed
- [ ] Make clear anything else important that changed
- [ ] Make clear current limitations/issues
  - [ ] No support for Pi3 or Pi0W Bluetooth
  - [ ] No Spotify connect (still)
  - [ ] Spotify playlists issues...?
- [ ] Include clear list of audio cards that are known to work
- [ ] Remove MPDroid recommendation? It's not very compatible with Mopidy any more (https://github.com/mopidy/mopidy/issues/1314).

ivr588, 2019-11-24:
Hi,

I'm using v0.7rc7
Can there be some documentation on how to use the backends, eg tunein?
I know what tunein is but no idea how to use it in a pimusicbox context and there's nothing addressing this in the manual.

thanks

kingosticks, 2019-11-24:
Hi, make sure it's enabled in the settings and then you should be able to select tunein for searching and it should also be present in the top level of the browsing page. Some backends also will provide playlists, tunein does not.

ivr588, 2019-11-24:
ah so that's how it works! many thanks