pimusicbox/pimusicbox #406: wifi password gets reset to "musicbox" sometimes

rainerda, 2017-04-04:
Hi,
this happened at least two times. The wifi password got reset to musicbox. I'm not sure how to reproduce it. One time I was adding the soundcloud info and the time before I was setting an ssh root password to "musicbox". After saving the musicbox became unreachable. When I saw the config, the wifi password was "musicbox".

rainerda, 2017-04-04:
as a workaround i can make playlists for all channels using my audioaddict api key. this way i would get classicalradio too. I already opened an issue for that in the mopidy-auddioaddict github page.

kingosticks, 2017-04-04:
This is interesting and very odd. I'll see if I can work out what's going on but if you find steps to reproduce it reliably that'd be very helpful. I assume you're changing these settings using websettings (rather than editing settings.ini in your text editor).

rainerda, 2017-04-04:
It was the chrome password autofill and me being stupid. Sry.

closed 2017-04-04 by rainerda

kingosticks, 2017-04-04:
No worries. I'm just glad as I was stumped!