pimusicbox/pimusicbox #407: No audioaddict playback

rainerda, 2017-04-04:
I don't get any sound when selecting an audioaddict channel after browsing audioaddict. I'm a pro user and the password is correct. The channel pops up in the UI though and can be paused and resumed to no avail.
I tried changing the bitrates, nothing happens.
Can I give you some better info? Where should I look in the logs?

kingosticks, 2017-04-04:
Any errors would be in /var/log/mopidy/mopidy.log

rainerda, 2017-04-04:
Stupid me and the chrome autofill settings. I found "musicbox" in my chrome passwords.

rainerda, 2017-04-04:
can't see an error there. As a workaround I made m3u playlists with all audioaddict channels.

kingosticks, 2017-04-04:
I think I must be missing something, why does creating a playlist work, what's the difference?

rainerda, 2017-04-06:
Good question. 
When I start playback with Browse -> audioaddict -> digitally imported -> Deep Techno
then I get no sound, same with JazzRadio and so on.
But playing the direct link (with my listener id from di.fm) works: http://listen.di.fm/premium/deeptech.pls?listen_key=[mylistenkey]
So I made myself a playlist of all streams that I got from the website and put them in a new playlist in /music/playlists.

kingosticks, 2017-04-06:
it would really help if we could see the INFO level logging. If you could run `sed -i 's/WARNING/INFO/' /etc/mopidy/logging.conf` and restart Mopidy (or reboot) then hopefully we can see what's going on. I guess it's not recognising you as a premium subscriber, perhaps the API changed (last update to the extension was 2 years ago).

rainerda, 2017-04-06:
I raised the logging, now I can see:
`2017-04-06 17:39:40,714 - ERROR    The stream is of a different type than handled by this element.`
my quality settings:
quality = 128k
i think they are using an aac stream for this bandwidth. maybe this has changed?

Skybeat, 2017-04-23:
I have exactly the same issue with the same error message like @rainerda

beshur, 2019-08-05:
The API has changed.

As a workaround, you can use the playlist links if you are a premium user.