pimusicbox/pimusicbox #408: Request for Bluetooth pairing as alternative to wifi

FailedFace, 2017-04-04:
Not sure if this is in the scope of the project but pairing to the raspi from a phone would provide a much desired ease of connect-ability without a requirement to connect to the network.

kingosticks, 2017-04-04:
Yes, that'd be good. Although I don't know if you can authorise the pairing on the musicbox side without any input. Perhaps you can force it to accept all connections but I don't think I'd want that running on my home network.  

I was also looking at something like [this](http://hackaday.com/2016/03/27/turn-your-rpi-3-into-a-ble-beacon/) the other day to advertise the system. But it's only really interesting when you don't have zeroconf/bonjour available (Android systems).

FailedFace, 2017-04-07:
Right, I don't think I would allow pairing without input either. I was going to add a touch display to my box, so automatic pairing wouldn't be an issue. 

Really, it's just another option it's not like it's really providing an easier interface for streaming the music to the pi. Just thinking about making this much more mobile, i was looking to make mine a box to lug around.

mon7gomery, 2017-05-25:
My current solution for mobile use:

I'm working on a mobile jessie based version of pimusicbox at the moment and going to use something like [this](https://gist.github.com/mon7gomery/209244c9f011ee7a6b7ff75159ad2e97)
Basically it's a bootstrapper that trys to connect to known wifis. If there is no known wifi available it will create an AP to access the rbpi. 
I added a GPIO toggle switch to choose the mode and a LED for status indication.

Example:
at boot: search for known wifi ssids, if non is available a AP will be created
wifi AP is active -(toggle switch)> search for kown wifis and connect, if non is available create an AP again
wifi client mode -(toggle switch)> create an AP

At the moment, this would only be usefull, if your going to use pimusicbox in "offline" mode (only lokal files).

Next step would be to add new SSID instead of replacing them. I'm not sure, but i think it is done that way at the moment.

To create a mobile online pimusicbox i think there are two options:
1) add an 3g usb modem
2) use your smartphone to create a hotspot and let pimusicbox connect to it

Has anybody made progress on the SD Cars corruption issues? 
As said before I'm using raspbian jessie and there weren't any corruption issues till now. But I had no deeped search for what is actually causing the sd corruption so far.

FailedFace, 2017-08-08:
I had another idea about the bluetooth idea. We could use NFC tags for the
pairing. That way it requires local physical access to rasp and a bit
better security. The protocol is SSP (Secure Simple Pairing). It requires
bluetooth 2.1 or higher.

I was playing with NFC tags when I came across it.

Bryan