pimusicbox/pimusicbox #409: Airplay not working

nean-and-i, 2017-04-09:
Airplay not working:
Unable to connect to Musicbox

root@MusicBox:~# shairport-sync -vvv
Looking for the configuration file "/etc/shairport-sync.conf".
Looking for configuration file at full path "/etc/shairport-sync.conf"
Output device name is "default".
The processor is running little-endian.
Version: "3.0-OpenSSL-Avahi-ALSA-metadata-sysconfdir:/etc"
statistics_requester status is 0.
daemon status is 0.
rtsp listening port is 5000.
udp base port is 6001.
udp port range is 100.
Shairport Sync player name is "MusicBox".
Audio Output name is "(null)".
on-start action is "/usr/bin/mpc stop".
on-stop action is "(null)".
wait-cmd status is 1.
mdns backend "(null)".
userSuppliedLatency is 0.
AirPlayLatency is -1.
iTunesLatency is -1.
forkedDaapdLatency is -1.
stuffing option is "0" (0-basic, 1-soxr).
resync time is 0.050000 seconds.
allow a session to be interrupted: 0.
busy timeout time is 120.
drift tolerance is 0.010000 seconds.
password is "(null)".
ignore_volume_control is 0.
volume_max_db is not set
playback_mode is 0 (0-stereo, 1-mono, 1-reverse_stereo, 2-both_left, 3-both_right).
disable_synchronization is 0.
use_mmap_if_available is 1.
output_rate is 44100.
output_format is 3 (0-unknown, 1-S8, 2-U8, 3-S16, 4-S24, 5-S24_3LE, 6-S24_3BE, 7-S32).
audio backend desired buffer length is 0.150000 seconds.
audio backend latency offset is 0.000000 seconds.
volume range in dB (zero means use the range specified by the mixer): 0.
zeroconf regtype is "_raop._tcp".
decoders_supported field is 1.
use_apple_decoder is 0.
no special mdns service interface was requested.
configuration file name "/etc/shairport-sync.conf" resolves to "/etc/shairport-sync.conf".
metdata enabled is 0.
metadata pipename is "(null)".
metadata socket address is "(null)" port 0.
metadata socket packet size is "500".
get-coverart is 0.
Successful Startup
avahi: avahi_register.
avahi: register_service.
avahi: service 'E3A17DD59D85@MusicBox' group is not yet commited.
avahi: request to add "_raop._tcp" service without metadata
avahi: service 'E3A17DD59D85@MusicBox' group is registering.
avahi: service 'E3A17DD59D85@MusicBox' successfully added.

kingosticks, 2017-04-09:
Is it this? https://github.com/pimusicbox/pimusicbox/issues/401

kingosticks, 2017-04-09:
Or more specifically, this? https://github.com/pimusicbox/pimusicbox/issues/398#issuecomment-289018725

nean-and-i, 2017-04-09:
yes it was the firewall, please use following iptables rules/files to fix it:

/etc/firewall/musicbox_iptables
# MusicBox Firewall
*filter
:INPUT DROP [1:72]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [15:2194]
-A INPUT -i lo -j ACCEPT
-A INPUT -p icmp -j ACCEPT
-A INPUT -p tcp -m multiport --dports 22,80,139,443,445,554 -j ACCEPT
-A INPUT -p udp -m multiport --dports 137,138,554 -j ACCEPT
-A INPUT -p tcp -m multiport --dports 5000:5005,6600,3689,38888,6680,49014,45962,49152,56278 -j ACCEPT
-A INPUT -p udp -m multiport --dports 6000:6005,5353,49152,60458 -j ACCEPT
-A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT
#avahi/bonjour: 5353 UDP
# ssh 22
#daap: 3689 TCP
#samba 137, 138 UDP (netbios-ns), 139 (TCP, smb) 445 (TCP, microsoft domainservice)
#webserver 80, 6680
#mpd 6600
# 554 rtsp, airplay
# 5000-5005 tcp, 6000-6005 udp: shairport
# 49152 udp: DLNA

COMMIT

nean-and-i, 2017-04-09:
or disable (comment first line) in interface ifup script _/etc/network/if-up.d/iptables:iptables-restore_ 

grep -r musicbox_iptables /etc/*
/etc/network/if-up.d/iptables:iptables-restore < /etc/firewall/musicbox_iptables

ReckeDJ, 2017-04-10:
Having the same issue here on Pi2. Is actually not an firewall issue, after restarting the musicbox the shairport-sync does not work. When I restart the service shairport-sync after initial loading, it works directly.

nean-and-i, 2017-04-10:
Good to know.
In my setup it was definitely the firewall, in my debug rule I saw incoming packets being dropped, while Itunes client was trying to connect, after a timeout I got an error message.
Without firewall or with the extended rules it worked immediately without touching shairport-sync  service.
Also it seems behaviour is depending on ios/macos version.

kingosticks, 2017-04-11:
@ReckeDJ It sounds like https://github.com/pimusicbox/pimusicbox/issues/398

ReckeDJ, 2017-04-11:
@kingosticks Yeah, I tried that one too, but the differences are that the musicbox is displayed in their audio output devices list. On my issue, after a reboot, it does not show up at all.. until I restart the shairport-sync service manually.
The weird thing is that the 'Discovery - Bonjour Browser' app on my iPhone finds the 'Remote Audio Output Protocol (AirTunes)', but when I request the details for it, it throws an error. After restarting the shairport-sync service or while starting the program with -vvv; it works immediately and I can request the details and stream music properly.
Yesterday I tried to debug/log the first time that the service is starting, but unfortunately I could not figure it out.

kingosticks, 2017-04-11:
Could you try changing `udp_port_range = 100` in /etc/shairport-sync.conf ?

ReckeDJ, 2017-04-11:
@kingosticks 
Thanks, but already tried that, with '3' and '100'!

kingosticks, 2017-04-11:
If you set `log_verbosity = 3` in /etc/shairport-sync.conf and then add `*.* /var/log/messages` to the bottom of /etc/syslog.conf you should get the shairport-debug logging (along with everything else).

ReckeDJ, 2017-04-11:
@kingosticks 
Thanks! Will try that at home ;)

kingosticks, 2017-04-11:
And one more possible experiment: edit (and uncomment) `mdns_backend = "external-avahi";` in /etc/shairport-sync.conf/

ReckeDJ, 2017-04-11:
@kingosticks Will try all that when I am home, but the wifi thing seems reasonable.

ReckeDJ, 2017-04-11:
@kingosticks 
Yeah! Tried your theory about the networking first and guess what? It worked immediately after a reboot. Simple by moving the upnp, shairport and monit starting after the network wait-loop. Could you create a fix for it? Thanks!

kingosticks, 2017-04-11:
Oh brilliant. Thanks for testing that out. I shall indeed sort that out sharpish.

gdgrosse, 2017-04-20:
Fantastic news. You guys are good. Thanks for fixing this. I just disabled the firewall on my end in my RC4 manually.

I happened to install PMB for the first time earlier this week, apparently after this bug was introduced. I could Airplay from any of my iOS devices and OS X, but not from any of my [many] android clients. I never thought to look at the firewall since the website states "Do not run it outside a firewall", so I assumed there was no firewall on the PMB. I literally spent the last 4 days messing with my 20 page long iptables in my DD-WRT router, thinking it was due to poor implementation of multicasting on my many subnets. Thank god I stumbled on this browsing through your commits.

Keep up the great work on PMB!

Yenn Time

kingosticks, 2017-04-20:
@gdgrosse what android client do you use? I could really do with one for testing.

gdgrosse, 2017-04-20:
AirAudio, AllConnect, AllCast and AllStream.

Nexus 6 running Android 7.1.1 rooted, Samsung Tab 10 Pro (SM-T520) running Android 4.2.2 not rooted but plan to root for some other testing soon, and a Samsung Note 4 running Android 6.0.1, not rooted.

Just let me know what you need.

collinjlesko, 2017-05-07:
The ip-table fix worked for me. Was not in Release Candidate 4, and AirPlay was failing to connect.

@nean-and-i

closed 2018-04-01 by nean-and-i