pimusicbox/pimusicbox #412: Lateset image - mopidy fails to start

plimptm, 2017-04-10:
I've tried running the latest 0.70 image and am facing the same issue after several reflashes.

The mopidy binary will not run properly - my /var/log/musicbox_startup.log terminates with this message:

`Traceback (most recent call last):
  File "/usr/local/bin/mopidy", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/mopidy/__main__.py", line 82, in main
    args.config_overrides)
  File "/usr/local/lib/python2.7/dist-packages/mopidy/config/__init__.py", line 80, in load
    raw_config = _load(files, defaults, keyring.fetch() + (overrides or []))
  File "/usr/local/lib/python2.7/dist-packages/mopidy/config/__init__.py", line 127, in _load
    parser.readfp(io.BytesIO(default))
  File "/usr/lib/python2.7/ConfigParser.py", line 324, in readfp
    self._read(fp, filename)
  File "/usr/lib/python2.7/ConfigParser.py", line 512, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
ConfigParser.MissingSectionHeaderError: File contains no section headers.
file: <???>, line: 8
'msgid ""\n'`

The same message is produced when trying to run mopidy (as root) from the terminal. The only modifications I made to the settings.ini file is to enable sshd.

Any ideas what's going on? I'm running an older B+ model pi.

kingosticks, 2017-04-10:
The error message suggests the settings.ini file is messed up but that seems pretty unlikely if you tried several times. What program did you use to make the changes, some text editors manage to screw up the line endings, for example.

plimptm, 2017-04-10:
Thanks for the followup. I used nano to edit the file. To check, I just tried replacing the settings.ini file altogether with a fresh copy from the repo, unmodified, and it is still failing to run the mopidy binary, same issue. I had to inspect the logs after shutting the pi down because I didn't have ssh access obvioulsy.