pimusicbox/pimusicbox #413: Audio streaming

anttiautio, 2017-04-13:
Hi, I get very choppy audio when trying to tune in to my favourite channel www.sleepbot.com using Pi Music Box. I wonder if there is anything I could do to fix this?

kingosticks, 2017-04-13:
What is your setup? 
* what model pi
* wired/wireless?

anttiautio, 2017-04-13:
I have Raspberry Pi 3 model B, Hifiberry DAC+, Pi Music Box 0.7.0RC4. Connection is wired, Spotify works fine.

anttiautio, 2017-04-13:
Probably not a connectivity issue since the station plays fine with VLC on my Mac in the same network.

kingosticks, 2017-04-13:
Are you using the same wireless module on your Mac?

kingosticks, 2017-04-13:
Ignore my previous (now deleted) post, crossed wires.

Is it just choppy or does it frequently stop? There is an ongoing buffering issue. Note that the buffering in Mopidy behaves very differently to VLC etc.

anttiautio, 2017-04-13:
It is like very rapid stutter -- the playback is interrupted very briefly several times per second. Might be the buffering issue you mentioned.

kingosticks, 2017-04-13:
Hmm I'm not sure, the buffering issue we've seen before usually has a few seconds of silence as it takes a while to refill the buffer back up to 100%. I'll try that particular station myself and see if I can get anywhere, might not get a chance for a couple of days though.

jcgruenhage, 2017-07-21:
> the buffering issue we've seen before usually has a few seconds of silence as it takes a while to refill the buffer back up to 100%.

Where can I find that issue? It seems that I experience that when listening to radio stations. Stops for a few seconds and then it continues to play.

I also have a problem with streaming music from Google Play Music, the music sometimes just randomly stops to play.

kingosticks, 2017-07-21:
> Where can I find that issue? It seems that I experience that when listening to radio stations. Stops for a few seconds and then it continues to play.

https://github.com/mopidy/mopidy/issues/1280

> I also have a problem with streaming music from Google Play Music, the music sometimes just randomly stops to play.

https://github.com/mopidy/mopidy-gmusic/issues/161