pimusicbox/pimusicbox #414: Youtube songs not playing (stuck on "sync")

Zehirlolz, 2017-04-14:
Hello,

I'm encountering a problem with the youtube streaming.
When I try to add a new song (by adding a stream) the song appears in "Now Playing" but does not start at all, (even after wating a long time). It's stuck on "sync" (Please see the screenshot bellow).

I try to add my songs by typing yt:youtube URL or with the uri I get when I click "Get currently playing" button in streams.  

ie : yt:https://www.youtube.com/watch?v=u9Dg-g7t2l4
or : youtube:video/Disturbed - The Sound Of Silence Official Music Video.u9Dg-g7t2l4

[image]

I am doing something wrong or is there a probleme with mopidy-youtube ?

Best regards

PS : It's a fresh install I only enabled spotify (which does work well), soundcloud and Youtube

jcass77, 2017-04-15:
Please take a look at the mopidy log files for clues and post the results here - there is some good [troubleshooting documentation](https://docs.mopidy.com/en/latest/troubleshooting/) available to help you get started.

kingosticks, 2017-04-15:
You'll need to edit /etc/mopidy/logging.conf and change the `level = INFO` lines to `level = DEBUG`. You'll then find the log at  
/var/log/mopidy/mopidy.log

Or you can replace it with the one at https://github.com/pimusicbox/pimusicbox/blob/develop/filechanges/etc/mopidy/logging-debug.conf and then find the log at /tmp/mopidy-debug.log. (I recommend this option).

Zehirlolz, 2017-04-18:
Hello,

Thanks for your answer, after doing what you asked me to, and looked at the logs it appears that it sounds like a plugin is missing : 

`
[...]
2017-04-18 12:42:03,052 - WARNING  There is no codec present that can handle the stream's type.
2017-04-18 12:42:03,054 - ERROR    Your GStreamer installation is missing a plug-in.
2017-04-18 12:42:08,435 - WARNING  Could not find a audio/x-unknown decoder to handle media.
2017-04-18 12:42:08,438 - WARNING  There is no codec present that can handle the stream's type.
2017-04-18 12:42:08,439 - ERROR    Your GStreamer installation is missing a plug-in.
2017-04-18 12:42:45,465 - WARNING  Could not find a audio/x-unknown decoder to handle media.
2017-04-18 12:42:45,468 - WARNING  There is no codec present that can handle the stream's type.
2017-04-18 12:42:45,469 - ERROR    Your GStreamer installation is missing a plug-in.
2017-04-18 12:43:19,825 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2017-04-18 12:43:20,635 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2017-04-18 12:43:22,353 - WARNING  /usr/local/lib/python2.7/dist-packages/mopidy/audio/actor.py:714: Warning: /usr/src/packages/BUILD/glib2.0-2.40.0/./gobject/gsignal.c:2580: instance '0x682b9558' has no handler with id '8407'
  result = self._playbin.set_state(state)
[...]
`

Am i supposed to install it manually or is there a problem with my setup ?

Thank you for your time !
Regards,

kingosticks, 2017-04-18:
A star as I know there's nothing you can do about that. Some (not all) youtube videos require a specific plugin that's not available in Raspbian Wheezy. Until the next release (where we move to Raspbian Jessie), we are a bit stuck with that. It would be better if that sort of error could be communicated properly without resorting to looking at debug logs, thats something we could look at.

Zehirlolz, 2017-04-18:
Thanks for your responsiveness :)

Would it be possible to enable jessie's repository to install theses plugins ? Might this works ?

If not i'll wait your the next release. thank you again for your support !

PS : Just to know, Is it possible to enable some form of authentification (using htpasswd for example) to restrein the access to the webinterface on a entreprise local network ? :)

Best regards !

kingosticks, 2017-04-18:
I don't know, that usually ends in a mess and it's a bit complicated as you ultimately want to install and use gstreamer1.0 to get the newer plugins and to do that you need to use the latest version of Mopidy (2.x). It's not trivial. 

There is no password support right now. Again, that's slated for the next release.

jolny, 2017-05-19:
I actually did a fresh install of Jessie, Mopidy 2.0, Musicbox-webclient, Mopidy-youtube etc and still got a similar issue. I haven't debugged it yet, though. Do you know which plugin it is that's missing, @kingosticks ?

kingosticks, 2017-05-28:
@jolny, with Gstreamer 1.0 in Mopidy 2.0? No. You should be able to see some errors in the logging and work it out from there.

But with Gstreamer 0.10 is was [webm causing the problem](https://github.com/pimusicbox/pimusicbox/blob/develop/create_musicbox0.7.sh#L118).