pimusicbox/pimusicbox #416: Issues with Soundcloud

Skybeat, 2017-04-23:
Hi,
I use v 0.7.0 RC4 and have some issues with SoundCloud.
At the soundcloude page i have some followings and likes for music, but if i try to see them in MusicBox (go to Browse -> SoundCloud->Following i reseive the follow error (in the log of MusicBox):

> 2017-04-23 20:47:34,053 - ERROR    SoundCloudBackend backend caused an exception.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 19, in _backend_error_handling
    yield
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 112, in _browse
    result = backend.library.browse(uri).get()
  File "/usr/local/lib/python2.7/dist-packages/pykka/threading.py", line 52, in get
    compat.reraise(*self._data['exc_info'])
  File "/usr/local/lib/python2.7/dist-packages/pykka/compat.py", line 12, in reraise
    exec('raise tp, value, tb')
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
    response = self._handle_receive(message)
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
    return callee(*message['args'], **message['kwargs'])
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 147, in browse
    return self.list_user_follows()
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 89, in list_user_follows
    for (name, user_id) in self.backend.remote.get_followings():
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 148, in get_followings
    name = playlist.get('username')
AttributeError: 'unicode' object has no attribute 'get'

If i go to Liked i can see the follow error in the log:

> 2017-04-23 20:51:07,624 - ERROR    SoundCloudBackend backend caused an exception.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 19, in _backend_error_handling
    yield
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 112, in _browse
    result = backend.library.browse(uri).get()
  File "/usr/local/lib/python2.7/dist-packages/pykka/threading.py", line 52, in get
    compat.reraise(*self._data['exc_info'])
  File "/usr/local/lib/python2.7/dist-packages/pykka/compat.py", line 12, in reraise
    exec('raise tp, value, tb')
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
    response = self._handle_receive(message)
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
    return callee(*message['args'], **message['kwargs'])
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 166, in browse
    return self.list_liked()
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 74, in list_liked
    for data in self.backend.remote.get_user_liked():
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 177, in get_user_liked
    liked = self._get('e1/me/likes.json?limit=1000')
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 235, in _get
    res.raise_for_status()
  File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 909, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 500 Server Error: Internal Server Error for url: https://api.soundcloud.com/e1/me/likes.json?limit=1000&client_id=93e33e327fd8a9b77becd179652272e2

The only thing i can do is go to Streams in Soundcloud. There i can see currently one title and if i start this one i have the sound but in the log file of musicbox i receive this warning:

> 2017-04-23 20:52:33,291 - WARNING  Element doesn't implement handling of this stream. Please file a bug.

I hope this will help you to find and fix the issue.

br,
Nico

jcass77, 2018-04-11:
Looks like an exception raised in the [SoundCloud backend](https://github.com/mopidy/mopidy-soundcloud). This issue might get more attention if logged in that repository.

closed 2018-04-11 by jcass77

kingosticks, 2018-04-11:
As it happens, I'm working on fixing up the soundcloud extension at the moment. I'm just waiting to finish that up before cutting a new pimusicbox release.