pimusicbox/pimusicbox #417: wireless adapter TL-WN722N not working

nean-and-i, 2017-04-26:
it would be great if common wireless adapters such as TP-LINK TL-WN722N are working ootb.
this would add high-gain wireless reception and much better streaming performance.

couldn't get it up and working for Raspberry 1.
Is the underlying debian version or kernel modules are outdated for this purpose?

kingosticks, 2017-04-26:
What driver does it require?

nean-and-i, 2017-04-28:
according to their linux support package it seems to be a "RTL8188EUS" chip 

Driver:
http://static.tp-link.com/TL-WN722N(EU)_V2_161112_Linux.zip

dmesg from ubuntu 16.04 (notebook)
[16700.940203] usb 2-1: new high-speed USB device number 4 using xhci_hcd
[16701.068603] usb 2-1: New USB device found, idVendor=2357, idProduct=010c
[16701.068607] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16701.068610] usb 2-1: Product: 802.11n NIC
[16701.068612] usb 2-1: Manufacturer: Realtek
[16701.068614] usb 2-1: SerialNumber: 00E04C0001

MaStr, 2017-05-07:
it mostly depends on the age of the USB stick, tbh. .. older TLWN722N are using kernel native drivers.

BTW: with newer 4.x kernels, the needed Realtek drivers are included. On our archlinux based PirateBox image, the drivers work out of the box... even in AP mode.

nean-and-i, 2017-05-07:
@MaStr have you proofed it?
pls. don't mix it up  with the older hardware model!
as of now, tp-link ship this model with new and different chipset "RTL8188EUS" and not as month ago with the highly compatible atheros chipset.
So, this one is definitely not work with newer kernels 4.x, also the correct Realtek firmware is missing, so no proper drivers included for most common distros!

because drivers are not available, this one does not work at:
- Raspberrypi 1 - Debian wheezy - FAIL
- Raspberrypi 3 - Debian jessie - FAIL
- Dell E74400 - Ubuntu 16.04 LTS - FAIL
- HP Elitebook - Fedora 25

pls. stop trolling "at archlinux everything is working", it is not helping in this case, we need to get it working for pimusicbox.

thanks

MaStr, 2017-05-07:
Hi,
First of all, I want to apologize. It was not my intention to troll around. I found the fact above per accident while debugging a udev rule for Piratebox startup.
My intention with posting that was, that you need to be careful about WN722N because different versions exists on the market. It seems we now know 3 different versions ...

My WN722N is the one of the first models which works out of the box since 3.x.

For Piratebox, we used to ship prebuilt packages on the image which are installed/changed during detection of devices.

I can try to get one of the problematic versions and see what is wrong .. if you like to. Possibly, I will stumble of that stick on the Piratebox project again...

Best regards Matthias

nean-and-i, 2017-05-07:
thanks, I appreciate your collaboration.

as far as I can see the pimusicbox is based on raspbian wheezy, so we'd need to have the driver source from tp-link compiled on raspbian wheezy.  driver source can be found here http://static.tp-link.com/TL-WN722N(EU)_V2_161112_Linux.zip 
Unfortunately I wasn't able to compile it at least on ubuntu 16.04 .

MaStr, 2017-05-07:
I did some further researches on this topic and additional preparations for cross compiling. 

I found out, that on the https://github.com/raspberrypi/linux the 8188EUS driver is included, but the vendor ID 0x2357, productid 0x010c  is not set in  

     linux/drivers/staging/rtl8188eu/os_dep/usb_intf.c

As PiMusikbox switched to v4.9.16 kernel using the source https://github.com/Hexxeh/rpi-firmware , the linked kernel source from TPLink might  be not valid anymore.
In an alternate driver source https://github.com/lwfinger/rtl8188eu it is possible to generate a dkms module and it already contains the needed driver changes for that wifi adapter https://github.com/lwfinger/rtl8188eu/commit/014ae2639386201fed92f98c971796602624b7ea

So probably prebuilding a dkms module with that alternate source is an option before stripping down the image back to release size (assuming I understood the build path of pimusicbox)

MaStr, 2017-05-09:
Crap.. the new TP-Link TL-WN722N I received today is "only" version 1.1 and has an ath9k chipset:

    usb 1-1.2: new high-speed USB device number 6 using ehci-pci
    usb 1-1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
    usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
    ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
    ath9k_htc 1-1.2:1.0: ath9k_htc: FW Version: 1.4
    ath9k_htc 1-1.2:1.0: FW RMW support: On

Trying to get a 2.0 somewhere...

nean-and-i, 2017-05-11:
damnn, I've bought mine at amazon.de.
we can swap or I can send you mine if you like?

MaStr, 2017-05-11:
I ordered mine on Amazon.de, too.. lol 
Yes, I would like to swap it. Can you e-mail me your address at matthias.strubel(at)aod-rpg.de and I'll send you over the stick (and after it arrives, you can send yours back. 
Thank you! 

I recognized your commit at RPI/Linux repository-fork. Do you think that patch will be integrated to the RPi-Kernel?

nean-and-i, 2017-05-11:
just sent you my postal address.

regarding the RPI/Linux commit: well, lets call it endless naivity,...
if this solves the problem, it should be implemented.
tbh. I haven't time to look into this in more detail.

thanks

mtdnet, 2017-07-23:
Hi@all,
i need the TL-WN722N V2 driver for kali on raspberry.
Anyone of us know how to compile it?
Some howto's that i found does not work...

Thanks

MaStr, 2017-07-23:
I was able to compile the driver, but I never got the device usable by the networking stack 👎

nean-and-i, 2017-07-24:
Hi @mtdnet , due to the "outdated" kernel and underlying distro non of the TL-WN722N versions will run ootb on pimusicbox.
I wasn't able to get v2 nor the v1 (special thanks to @MaStr , who exchanged his v1 with my v2), in the end I was forced to switch to https://volumio.org  where at least v1 worked ootb.
I don't see a chance that v2 will work in the near future on pimusicbox as the whole underlying distro needs to be updated to a recent version plus some extra special tweaking for the TL-WN722N v2. Be aware that the v2 doesn't even run ootb on current Ubuntu 16.04 LTS.

kingosticks, 2017-07-24:
Is it actually a kernel issue or a missing non-free driver that's required? 0.7Rc4 has kernel 4.9.16 and 0.7rc5 will have the latest as of last week. If it'd just missing firmware that should be feasible.

kingosticks, 2017-07-24:
I'd like to help get this working but I'm not sure what needs to be done. Is is still the 8188EUS that's got the issue? Does it work on any version of debian?

kingosticks, 2017-07-24:
Has anyone tried configuring wpa_supplicant.conf to use wext as described in http://www.thelinuxrain.com/articles/getting-realtek-8188eu-wireless-adapters-to-work-in-linux-and-possibly-other-wireless-realtek-chipsets ?

nean-and-i, 2017-07-24:
nope, that's not working, also I tried a lot with replacing certain firmware files etc,...
v2 is quite a pain, after investing some days to try to get it working on ubuntu I gave up, also I was not abler to compile the source from tp-link (they state it as driver).
But it seems it is working on arch linux based distros.

it could well be that v1 is working on newer kernel version, but the thing is that even on amazon you'll get unpredictable versions, one gets v1 while another get v2,...

Acidic9, 2017-07-27:
Here's my story: I bought the TP-LINK TL-WN722N Version 2 a couple of days ago. I am running Kali Linux (4.9.0-kali4-amd64) on my laptop (not in vm). I've spent the last two days trying to get the drivers to work for my wireless adapter and I managed to get it to show in iwconfig from [this installation here](https://github.com/mfruba/kernel/tree/master/drivers/TL-WN722N_v2.0-Ralink).

My problem now, is when I type `airmon-ng check kill; airmon-ng start wlan0` I get the following output:
```
Killing these processes:

  PID Name
  693 wpa_supplicant

PHY	Interface	Driver		Chipset

phy0	wlan0		??????		 
cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: No such file or directory

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
```

So many hours spent trying to get this shit to work :/

kingosticks, 2017-07-27:
@nean-and-i have you tried the http://fars-robotics.net/install-wifi script? There seems to be an available module for the kernel used in v0.7rc4 (4.9.16-v7+ #978) but I don't have the wireless stick so I can't test. If this works out I can include all these pre-built drivers.
```
wget http://fars-robotics.net/install-wifi -O /usr/bin/install-wifi && chmod +x /usr/bin/install-wifi
install-wifi
```

ankus031, 2017-07-29:
please anybody help me...i want to buy tp link wn722n wifi adapter... but now 3 versions are available.. please suggest me which version is able to monitoring mode and packet injection..an also support on kali linux on Vmware...please help

MaStr, 2017-07-29:
I think this is not the correct place to ask for instruction of "some" distribution. Just buy _a different_ wifi adapter. Maybe a Ralink RT5370 based USB-devices like this  http://www.amazon.com/exec/obidos/ASIN/B00H95C0A2/ .

Acidic9, 2017-07-29:
My friend hacked his neighbors wifi using V1. I believe V1 can be used as plug and play without any pre-setup on Kali. But whatever you do, don't buy V2... it will cause you days of headache with no result.

ankus031, 2017-07-29:
thank you..

please tell me a wifi adapter model which can perform packet injection & monitoring mode...in windows & kali linux via vmware.

 #MaStr i saw your suggestion but that device for ruspberry pi.. but i need for my laptop.

MaStr, 2017-07-29:
Linux = raspberryPi.
If it works on the pi , you can be sure it works under Linux.

But I think nobody can tell you if it will  work in the constellation VMware & windows .
For Qs about kali, you may ask in the kali forum .

nean-and-i, 2017-07-30:
@kingosticks , sorry I don't have the v2 anymore, also I'm very busy with other projects atm. , I'm open do donate a v2 stick for you to test (pls. send me a amazon wishlist link or something similar, I'll cover the costs)

@Acidic9 @ankus031 , I need to step in here; 
first -> wrong place, its about pimusicbox and how to get the TL-WN722N working, you're welcome to participate, but its not about kali here ;
second -> its about ethics, you know. you'll hardly find someone in a  linux community who is going to give you something like a out-of-the box solution if you're going to misuse it, like to hack someone else's wifi without permission, and probably without understand the technology.
If you're really interested into how to things work or to get things working there are wonderful projects out there you can learn from.
Don't waste your skills, better help @kingosticks eg. on testing and please participate to find a solution.
Last, one said, no one will help script kiddies who are going to cause chaos -> hacker ethic.
Especially in these times our common effort should be to save and protect "our/everyones" privacy!
help this neighbor to fix the weak wifi, not abuse it, you'll benefit from it ;-)
Thank you for your understanding.

Acidic9, 2017-07-31:
@nean-and-i It's purely for learning purposes. I'm even attempting to hack my own wifi. I've been trying for so long and it's something that must be done or else all hell will break lose.

kingosticks, 2017-08-01:
Thank you @nean-and-i, that's very kind. From the Amazon UK page it looks like a lot of people are complaining about receiving the v2 so I'll just order one from there. You can never have enough wifi dongles. If that doesn't work out then I might take you up on your offer (I'll pay for shipping, of course).

And thanks for taking the time to highlight what this repository is and isn't for, you are exactly correct. There are far more appropriate forums to discuss Kali Linux so please use those.

nean-and-i, 2017-08-02:
@kingosticks , no worries ;-)
have you thought about to place a donation button (paypal/amazon) on the project site, just for such purpose?

expl0iter99, 2017-10-18:
just to add something here, as per driver page of WN722N page, it will work only on older version or linux, so I installed older version of ubuntu with correct linux version and update header. and I was able to set v2 of this cr*p into monitor mode. with some modification in driver, I was able to run it on Kali version 2. but its not working on latest version of linux kernel, another thing I want to share is, v2 does not support packet injection (with the drivers).

kingosticks, 2017-10-18:
I'm not sure how to make this any clearer:

If you are using Kali Linux, this is not a relevant place to post.

Josue198s, 2017-10-22:
Here is a solution.. 

https://github.com/mfruba/kernel/tree/master/drivers/TL-WN722N_v2.0-Ralink

For Kali Linux worked.

safal2000, 2017-11-25:
I have install TL-W722N v2 in Kali 4.13.0 but it doesn't support monitor mode. Can any one help me out??
Or anyone wants to exchange it with V1?

nean-and-i, 2017-11-25:
@safal2000 ,  please consult kali community forum https://www.kali.org/community/ 

If you (and all others who refers to kali) are not using pimusic box, but kali linux, this is a very wrong place to report an issue.

This is an issue thread for pimusicbox and NOT kali linux, and it is not like that the libraries or even source easily work OOTB on both !

thank you

fhadiel, 2017-11-29:
@Josue198s  yeah it works, but i cant get into monitor mode

closed 2018-04-01 by nean-and-i

0E800, 2018-05-25:
@nean-and-i 
Your quote: 
> Last, one said, no one will help script kiddies who are going to **cause chaos -> hacker ethic.**

What you mean by that? Did you mean: hakuna matata ?

Shame on on the rest of you too. Like nunya never grep'd information for your own selfish purposes from a github project without even giving it a star let alone a push/pull request.

Upgrade your grey-matter folks, I mean: hakuna matata