pimusicbox/pimusicbox #418: Spotify - No tracks found....

MrGJones, 2017-05-03:
using v0.7.0 Release Candidate 4

when i browse spotify to featured playlists, genres & moods etc.. i get "no tracks found..." returned but when i search for tracks it yields results and can be played ok?

kingosticks, 2017-05-05:
Unfortunately this is due to Spotify's breaking change to their API which renders most/all of playlists owned by the spotify user incompatible with the libspotify library that mopidy-spotify uses. https://github.com/mopidy/mopidy-spotify/issues/114 would fix this.

MrGJones, 2017-05-05:
ok ill take a look into this.. many thanks for the reply!

geeflaxx, 2017-05-19:
a short question:
Do i need a spotify Premium account when i want listening music via pimusicbox/raspberrypi/bluetoothbox?

kingosticks, 2017-05-20:
You need a Spotify premium account when using Spotify on any non-Android arm device. So that includes Raspbian, which pimusicbox is built on. I've no idea what Bluetoothbox is.

BLX-S, 2017-06-25:
Is the fix already integrated in the PMB image? I read the issue about the Spotifyweb and mopidy-spotify merger but I didn't get if something already was done or not.

kingosticks, 2017-06-25:
No this fix isn't in 0.7 and I don't think it will be. Unless someone else
backports the fixes which is a lot of work and personally I don't think
it's worth the time.

pdnellius, 2018-03-13:
Has anyone found a way to get Spotify working? "Spotify browse" and "Spotify" searches both return `No tracks found...`

kingosticks, 2018-03-13:
There's a workaround described here you can apply: https://discourse.mopidy.com/t/spotify-search-don-t-work/2133/8

pdnellius, 2018-03-13:
@kingosticks , 

Thanks so much for the info. The search _acts_ like it's working and queues the song. The file even acts like it's starting to play and displays the correct time left. However, I'm getting this in the mopidy.log:
```
2018-03-13 12:35:28,593 - WARNING  Element doesn't implement handling of this stream. Please file a bug.```

Anybody seen this before?

kingosticks, 2018-03-13:
I actually did see this last week but only on my Pi ZeroW and I didn't get to the bottom of it. I think it started happening when I enabled librespot (Spotify Connect) but I do need to go back and verify that. Are you using librespot?

pdnellius, 2018-03-13:
@kingosticks , 

No, I have Spotify Connect disabled. Any other logs that might be helpful to get to the bottom of this?

kingosticks, 2018-03-13:
Your audio setup (`aplay -l`), your full mopidy config (`service mopidy run config`). You could try getting the gstreamer log too (https://docs.mopidy.com/en/latest/troubleshooting/#debugging-gstreamer). I'll have a play with my PiZW system tonight.