pimusicbox/pimusicbox #420: Monitor sshd with monit

MaStr, 2017-05-07:
Hi,
what do you think about handing the control over sshd to monit as well?

I just hit the issue, that the Pi run into issues and all daemons dropped including the ssh. I assume the root cause is an OOM issue.. but I can not verify it without SSH access, and this daemon is killed as well.

I suggest to move over a control-config file for ssh to monit's conf.d folder if SSH is enabled - or enable/disable it via monitrc commands.

What do you think?

best regards
Matthias

kingosticks, 2017-05-07:
I guess we could. I don't think we enable the monit server that allows you to control it although might be remembering that incorrectly. 

But it won't actually help with your problem. It'd only help if sshd (dropbear) crashes which I think we've had zero reports of.

MaStr, 2017-05-07:
Yes, it won't bring me down to my issue. I am assuming it is an issue with the wifi connection, but haven't time to track this down until now. 

Well, on musicbox 0.7 RC4  monit is enabled to control:

    $ ls -1 monit/conf.d/
      mopidy
    $ sudo cat monit/monitrc.d/mopidy 
      check process mopidy with pidfile /var/run/mopidy.pid
        start program = "/etc/init.d/mopidy start"
        stop  program = "/etc/init.d/mopidy stop"
        restart program = "/etc/init.d/mopidy restart"
        if failed port 6680 protocol http for 2 cycles then restart

You can monitor dropbear like the normal openssh-server, with a new service file. If you are interested in that, let me know... I can try to prepare a PR during next week.

kingosticks, 2017-05-07:
Apologies, I thought you meant using the monit cli to enable/disable the monitoring (which requires the monit http server to be running). To be honest since we don't have any actual issues with dropbear that require it to be restarted/monitored I'm not hugely interested. But if you feel it adds value and the change to startup is simple then we can merge a PR.

closed 2017-05-23 by MaStr