pimusicbox/pimusicbox #422: Making it available at music.box instead of musicbox.local

up2go-sschwarz, 2017-05-20:
Many Wifi-Routers are accessible through easy.box or fritz.box etc.
It would be great if the pi musicbox was available at music.box

I couldn't find the setting or piece of code in this repo, so I was not able to do it in my personal version.
I wonder if anyone can tell me, how to achieve this, or if there is someone willing to implement this for everybody.

MaStr, 2017-05-23:
Hi,
I have to disappoint you.

.local addresses are configured are resolved via mDNS, which is a server and (more important) client side setting. See https://en.wikipedia.org/wiki/.local & http://web.archive.org/web/20160608083415/http://avahi.org/wiki/AvahiAndUnicastDotLocal
So, you can change the avahi configuration to announce music.box , but you need to adjust every client to resolve. 

The other thing is the resolution of a normal domain name like .box or .lan. This name resolution is driven by your dhcp & dns server (most cases the Internet router). Every client receiving an IP address is registered at this device and sending over the client's hostname. 
The dhcp server takes this name and adds it to the name resolution list together with a search domain.

On fritzbox, you have the (search) domain "fritz.box". A client gets the FQDN client: client.fritz.box
In another case, you can run OpenWrt as router, which runs ".lan" as the search domain for the local network, so the client is known with the FQDN client.lan.

This setting is mostly related by your setup at home. Especially on FritzBox you are not able to add custom hostname entries, which will resolve another FQDN then *.fritz.box :-(

TL;DR It is not possible to influence the complete name in the network out of a client like pimusicbox

best regards Matthias

kingosticks, 2017-05-23:
Thanks for that explanation, sounds like we can close this.

closed 2017-05-23 by kingosticks