pimusicbox/pimusicbox #423: Make Wifi Channel 13 available

kingosticks, 2017-05-23:
From https://discuss.mopidy.com/t/wifi-channel-13/1634:

>I live in the UK where Wifi channel 13 is allowed. After a reset my router decided to switch to channel 13.
Unfortunately, out the box, PiMusicBox does not allow the wifi region to be set and therefore will not scan or connect to a wifi network on channel 13.
To work around this I installed the following packages from the raspbian packet repository:
wireless-regdb_2016.06.10-1_all.deb - http://archive.raspbian.org/raspbian/pool/main/w/wireless-regdb/
crda_3.13-1_armhf.deb - http://archive.raspbian.org/raspbian/pool/main/c/crda/
Also I edited /opt/musicbox/startup.sh to add country=GB to the generated /etc/wpa.conf file (line ~98).

```
if [ "$INI__network__wifi_network" != "" ]
then
    #put wifi settings for wpa roaming
    if [ "$INI__network__wifi_password" != "" ]
    then
        cat >/etc/wpa.conf <<EOF
            ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
            update_config=1
            country=GB
            network={
                ssid="$INI__network__wifi_network"
                psk="$INI__network__wifi_password"
                scan_ssid=1
            }
EOF
```

kingosticks, 2017-05-23:
Probably easiest to just add a new `wifi_country` setting to settings.ini (with a comment) and can add the two new packages to create_musicbox0.7.sh.

massycat, 2017-05-24:
Here is a pull request to allow this:
https://github.com/pimusicbox/pimusicbox/pull/424

kingosticks, 2017-05-25:
Merged in e6d4e14f873542de80d2f48df16f4544ebb9f9f7

closed 2017-05-25 by kingosticks