pimusicbox/pimusicbox #425: No Airplay

xthursdayx, 2017-05-26:
Hi, first, thanks for Pi Musicbox! After trying all of the other Pi music distro's I've found this one to be the best! My only issue is that I'm having trouble connect to the device with airplay/shairport. When I first installed Pi Musicbox I was able to stream with Airplay from my macbook for the first few days with no problem, though my wife tried was unable connect from her windows-based machine (using tuneblade). For some reason airplay then stopped working for me as well. I can see the Airplay device under my Mac's speaker output options, but I can't connect. I've since reformatted my SD card and reinstalled a fresh image of Musicbox, but still can't connect via Airplay. I also tried renaming the device from the standard "MusicBox" just in case there were settings stored on my Mac for this output device, but I still can't connect. Any ideas? Thanks!

kingosticks, 2017-05-26:
This sounds like #409, both the firewall blocking and the limited number of control ports.   You should be able to workaround it by following both parts of https://discuss.mopidy.com/t/after-first-reboot-cant-access-musicbox-local/1580/15

xthursdayx, 2017-05-26:
Great, thanks I'll try this out. Sorry, I didn't find that (very similar) issue (#409) in my googling, and I guess I only checked open issues here on github.

xthursdayx, 2017-05-26:
The instructions at https://discuss.mopidy.com/t/after-first-reboot-cant-access-musicbox-local/1580/15 worked for me and resolved the issue. 

I also tried the modified iptables rules at [#409 (comment)](https://github.com/pimusicbox/pimusicbox/issues/409#issuecomment-292794769) rather than disabling the firewall, and that also seemed to work.

kingosticks, 2017-05-26:
Thanks for updating.  I need to upload a new image with these fixes as a few people have run into it now.