pimusicbox/pimusicbox #426: Remove unused packages for 1GB SD-Card

afaenger, 2017-06-01:
I've managed to reduce the size of the latest musicbox_v0.7.0RC4 image so that it fits on a 1 GB sd card. The new image is roughly 860 MB and when loaded onto the sd card and resized to the full size 160 MB free space are available.

These are the steps to remove the unused packages (mainly command line and development packages)

apt-get --purge remove mc mc-data xserver-common x11-xkb-utils xkb-data libxkbfile1 dpkg-dev groff-base
apt-get --purge remove libaspell15 libhunspell-1.3-0
apt-get --purge remove man-db debian-reference-en debian-reference-common
apt-get --purge remove libicu48
apt-get --purge remove binutils cpp cpp-4.6 gcc-4.6-base gcc-4.7-base libc-dev-bin libc6-dev
apt-get --purge remove make m4 autotools-dev

rm -R /usr/share/man/*
rm -R /usr/share/doc

Maybe its possible to integrate this into the build script so that the new images can be directly loaded onto smaller sd cards.

kingosticks, 2017-06-01:
Yes this sounds good in theory. 

However, I have been toying with spotify-connect-web and it seems that relies on a rather large (150mb+) chroot environment in order to provide rpi1 support. That would put a pretty big nail in the 1GB SD card coffin. Personally, I think people would prefer connect support, you don't see sd cards in shops anymore and I've had no other comments about shrinking it back down. Im not sure what's best here.

afaenger, 2017-06-02:
I know that as development continues, it might not be possible to put everything on a 1 GB SD card anymore.

I still have a few 1GB SD cards in my drawer that cannot be used for any other purposes as they are so small... With this patch I don't need to get another bigger card for the current distribution of pimusicbox. Maybe others are in the same situation, so I thought I share my approach.

nightah, 2017-06-23:
@kingosticks has there been any thought into librespot as opposed to spotify-connect-web? If so, does it suffer from the same chroot env problem?

I've seen a fair few people pulling down pre-compiled binaries for librespot from: https://github.com/herrernst/librespot and installing alongside the current version of musicbox for spotify-connect support.

kingosticks, 2017-06-23:
You would have to install old versions of librespot as the latest ones
require a newer version of libc than what is available in wheezy. And older
versions are getting less and less likely to work since there have been a
few Spotify changes that have required librespot updates/fixes in the last
few weeks in particular.

So, 0.7 will probably get spotify-connect-web, but it'll have to be
downloaded and installed at runtime (rather than shipped) since the chroot
is massive. They already have a script to automate that.

But going forward it'll be librespot.

kingosticks, 2017-06-23:
And your answer there shows my info is very outdated. No reason to not use librespot now. I knew there was a good reason I hadn't released 0.7final yet!

commit 491a7205 (2017-07-12): Remove unrequired packages and shrink image (Fixes #426)
closed 2017-07-12 by kingosticks