pimusicbox/pimusicbox #428: Last.FM scrobbling not working

anttiautio, 2017-06-22:
Hi, when I play songs on my Pi Music Box they are displayed as "playing now" in Last.FM but are not recorded in song history. I run Pi Music Box 0.7.0 rc 4 in Raspberry Pi 3 with HifiBerry DAC+. Pleaase let me know if you need any further information.

aidanb, 2018-04-13:
I am having the same issue. If I play via Spotify (ie from the Spotify application), then it scrobbles. But playing a Google Play album via the MusicBox interface does not scrobble.

kingosticks, 2018-04-13:
Are you able to try another service or local file to verify if this is
specific to google music?

aidanb, 2018-04-13:
I just tested with Spotify via MusicBox (ie not using the Spotify
application) and it did not scrobble.

Links:

  1. https://github.com/pimusicbox/pimusicbox/issues/428#issuecomment-381074248
  2. https://github.com/

kingosticks, 2018-04-13:
Thanks. This is on my to do list for the weekend.

aidanb, 2018-04-13:
No worries. Let me know if there's anything else I can do to help.

Links:

  1. https://github.com/pimusicbox/pimusicbox/issues/428#issuecomment-381082377
  2. https://github.com/

kingosticks, 2018-04-13:
Can you confirm you are running v0.7.0RC6 (the latest release)?

aidanb, 2018-04-13:
It was the latest version when I installed it three weeks ago :)
Is there a way to check the version from the webapp? I can't see
it anywhere.

Links:

  1. https://github.com/pimusicbox/pimusicbox/issues/428#issuecomment-381086862
  2. https://github.com/

kingosticks, 2018-04-13:
The last version was released 25th March so you are probably good. One simple way to tell is by going to the websettings and checking the "Audio Output" drop down menu to see if there are entries for AudioInjector Stereo and AudioInjector Octo cards. It's also possible to read the version info from the file at /etc/issue but that's only possible if you log in (via ssh etc). But that's reminded me that I did want to [add that to the webclient somewhere](https://github.com/pimusicbox/mopidy-musicbox-webclient/issues/252) - thanks

aidanb, 2018-04-13:
I do indeed see AudioInjector Stereo and AudioInjector Octo.

Links:

  1. https://github.com/pimusicbox/mopidy-musicbox-webclient/issues/252
  2. https://github.com/pimusicbox/pimusicbox/issues/428#issuecomment-381146116
  3. https://github.com/

Ketjow3, 2019-02-06:
Last.fm still not working, I use latest build img (release candidate 7) on Rpi3B. A have correct login and password. I want use scrobble with playing local mp3 files. Please fix it.

kingosticks, 2019-02-06:
I am going to need you to [provide a debug log](http://docs.pimusicbox.com/en/latest/troubleshooting/). Thanks.

kingosticks, 2019-02-07:
Another user has since provided a debug log. This is not the same issue as the one originally reported, this one is new and much bigger. It doesn't look like the old version of Python available in v0.7.x is supported by the scrobbler extension's main dependency (pylast). This is not a simple fix and I am not going to spend my time on fixing it for v0.7.x. Time that I do spend on pimusicbox is better spent on the long, long, long awaited v1.0.0 release where this issue simply won't exist by virtue of using the latest python version.

underyx, 2019-03-02:
I worked around this by manually downgrading pylast with `pip install pylast<1.7`