pimusicbox/pimusicbox #432: Add support for Matrix MPD DJ

jcgruenhage, 2017-07-19:
Would it be possible for you to add [this cool project](https://github.com/Half-Shot/matrix-mpd-dj) to your image? It allows users to control MPD through [matrix](https://matrix.org).

kingosticks, 2017-07-19:
Sure but I have no idea what it is.  does it makes sense without a STREAMURL? Mopidy doesn't support streaming its audio via http and therefore nor do we.

jcgruenhage, 2017-07-19:
At least play/pause and stuff like that should work. Also, I think it does more than what the readme contains, from looking at the commits.

kingosticks, 2017-07-19:
Having looked a bit closer at what matrix is I'm not convinced this is something aimed at the pimusicbox crowd. If someone wants to provide a PR adding this software then that's great but I'm not going to look into it myself.