pimusicbox/pimusicbox #435: Firewall loaded after startup.sh

poikilotherm, 2017-08-08:
Hi there,

I'm currently testing Pi Musicbox 0.7 RC5  on a Rasberry Pi 2, connected via ethernet cable.

Most of the time I cannot connect to the Pi via SSH, sometimes it works. This is similar to #39 - iptables rules restored after the startup leading to filtered ssh traffic.

The hole thing about setting rules in startup.sh and restoring rules by if-up.d scripts seems to be prone to a race condition...

Relevant settings.ini parts:
```
enable_ssh = true
enable_firewall = false
wait_for_network = true
```

poikilotherm, 2017-08-08:
Things popin' up in my mind: wouldn't it be the easiest to extend the `/etc/network/if-up.d/iptables` script to honor settings.ini? That should avoid any race conditions and doesn't break current behaviour.

kingosticks, 2017-08-09:
This is what wait_for_network is supposed to avoid so it sounds like that isn't working. Does your startup log explain why that's the case? Is there a significant time different between when the local interface comes up and eth0/wlan0 and its getting a local ip address which breaks out the loop? 

Your suggestion sounds reasonable, or we could control ssh by starting and stopping the service (like the other optional services) rather than mucking around with firewall ports. I don't know what it's done differently, I see no reason to run an ssh server if you're not using it for remote connections.

commit 96105290 (2018-01-16): Don't block SSH port in firewall, just start the server as required. (Fixes #435)
closed 2018-01-16 by kingosticks