pimusicbox/pimusicbox #437: hifiberry DAC+ selection causes issues

wedgetail84, 2017-08-28:
I've just installed pimusicbox on my pi zero w with hifiberry dac+ zero 1.0.1. All is good and I can get into the menu on my mac and connect via airplay etc.. But when I change the output to the hifiberry dac+ setting and click apply (reboot) it goes wrong. On the monitor for the pi I can see it going round and round in cycles - every so often I see:

[FAIL] Stopping watchdog daemon... failed!
[FAIL] Starting watchdog keepalive daemon... failed!

with the FAIL and failed! in red. It's definitely the hifiberry setting that causes this as it's done it twice and turning the airplay on on its own is fine. 

Does anyone have ideas on how to get this working?

Thanks

kingosticks, 2017-08-29:
Hello, please try https://discuss.mopidy.com/t/boot-loop-enabling-dtoverlay-pi-dac/1816/10?u=kingosticks

wedgetail84, 2017-08-29:
Thanks kingosticks - that looks like it's me. I see the line you've put up - but what do I do with it, put it in the config file?

wedgetail84, 2017-08-29:
Thanks a lot - that solved it! However I now have a new problem, it doesn't seem to recognise the soundcard. I'll post a new thread on it as it's not relevant to this title...

kingosticks, 2017-10-04:
Stopped the reboot loop in https://github.com/pimusicbox/pimusicbox/commit/7babeb5c76ca87337259bd3964884aea260e1537