pimusicbox/pimusicbox #438: hifiberry dac "unable to find soundcard"

wedgetail84, 2017-08-29:
Having got pimusicbox to boot with the hifiberry dac plus option selected I can't get any sound out of it (it will come through HDMI from the pi). On the boot log I can see:
Device Tree Overlay 'hifiberry-dacplus' is already present in /boot/config.txt but unable to find soundcard.

Then in SSH when I run aplay -l I get:
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Not sure where to go from here...

wedgetail84, 2017-08-29:
hardware is pi zero w and hifiberry dac+ zero 1.0.1. I've double checked my solder and can't see any dry joints...

wedgetail84, 2017-08-31:
Just to close this - I selected the hifiberry dac option not dac+ and it all works great. I should have tried this before but I assumed a dac+ zero was closer to a dac+ than a dac...

kingosticks, 2017-08-31:
Interesting, ok, thanks for reporting back.

kingosticks, 2017-10-04:
I need to better document this in the websettings