pimusicbox/pimusicbox #439: spotify connect hangs up i think

moscher, 2017-08-29:
when the Musicbox is fresh started i can connect from my Tablet/ smartphone via spotify connect.

But when i stopped the connection and try a reconnect a few hours later i find no musicbox in my spotify-connect device list.
I had to reboot the musicbox than everthing runs fine again.

Maybe there is a fix outside??

greets
Falk

KryptusDev, 2017-08-29:
I am having similar issues. The music box sometimes takes a long time to appear in Spotify's device list or disappeares when skipping a song or listening on my phone.
One or two reboots usually do the trick though, after all, the feature is marked as buggy in the settings entry.

moscher, 2017-08-31:
Yeah the spotify-connect feature is marked as buggy.

I have "rasPI check" installed at my tablet, for a easy reboot.
But often rasPI Check cant connect to my musicbox via SSH. 
So i need to pull the USB cable to switch off power.

I think the spotify-connect feature hangs the complete system.