pimusicbox/pimusicbox #441: Raspberry Zero W - Bluetooth

tringler, 2017-08-31:
Hello,

unfortunately I'm unable to use the pi musicbox image with my Raspberry Zero W if I want to use the built-in bluetooth chip.

The bluetooth device is not working (not found at all), because I assume the raspbian image is too old. If I try to update to jessie or stretch the musicbox is not reachable anymore (jessie) or not booting up at all (stretch).

Any ideas how to solve it? Is anyone using a RPiZW with bluetooth successfully?

Thanks in Advance!

kingosticks, 2017-09-01:
That's a known issue listed in the changelog somewhere. I currently have no fix for that.

tringler, 2017-09-01:
Ok :( - So I need to attach an BT usb adapter right?

kingosticks, 2017-09-01:
Yep. Unless you find the solution for wheezy, of course!

closed 2017-09-02 by tringler

pierreboissinot, 2019-06-24:
Hi, is it fixed now ?

kingosticks, 2019-06-25:
Sorry, no. There won't be a fix until Musicbox v1.0. Whenever (if ever) that is. It is a little misleading I left this closed, I apologise for that.