pimusicbox/pimusicbox #447: Add interface for installing bluetooth speakers

JanFrancisty, 2017-10-20:
Hi,
Thanks for your work, musicbox is just an amazing platform. But I have a question or request, does it possible to add function to add bluetooth speakers ?

kingosticks, 2017-10-20:
As it stand currently, the Raspberry Pi onboard Bluetooth does not work in any of our v0.x releases. External bluetooth dongles do work (assuming they already have suitable driver support in Raspbian) but there is currently no support built into the Pi Musicbox software so you have to do everything manually (scanning, pairing etc).

We do want this feature, and it will be possible to add it once we upgrade to a newer version of Raspian. Another user has been kindly working on this very recently and from what I hear, has got much of it working to some degree. I'm currently working towards finishing the v0.x releases so I can get back to developing the the next release which will incorporate this bluetooth support.

roberts-gh, 2019-07-11:
oh PLEASE add this functionality, so that I can stream from the Pi to an existing BT receiver hooked into my stereo system!