pimusicbox/pimusicbox #449: possible to upgrade single modules?

huwgrind, 2017-10-20:
I used pip --upgrade to upgrade the mopidy-mopify module to 1.61 in hope of less crashes with spotify and spotify connect. But since then, I cant acess the web interface anymore.. So I will reinstall Pi Music Box. Was there a problem with the upgrade or is it just not possible to update single modules? 

Despite: Thanks a lot for the great work!

Regards, 
Hugo

kingosticks, 2017-10-20:
Upgrading modules isn't supported, see 
http://docs.pimusicbox.com/en/latest/faq/. Reinstalling is the best course of action once in a broken situation like this. 

A pip upgrade like that will update the package and all its dependencies. This updates Mopidy to the latest version which we (still) don't support. Mopidy-Mopify is very separate from Spotify-connect and since it's just a Mopidy frontend I don't think it'll help anyway. Plus RC5 should still have the latest versions of both librespot (for Spotify-connect) and mopidy-mopify.  

I'm not surprised to hear librespot crashing and there's not much we can do about that due to the nature of the software. But I am more surprised to hear of Mopidy crashing, can you enable the debug logging and provide the errors?

If you do want to upgrade just mopidy-mopify, you can run `pip install --upgrade --upgrade-strategy only-if-needed mopidy-mopify`. Note that this only works for packages that do not have a hard requirement on a more recent version of Mopidy. Unfortunately that varies randomly from package to package, hence it's easier to just not support it at all.

huwgrind, 2017-10-21:
I see, thanks for the explanation, I reflashed the 0.7RC5 zip now. The crash happened while using mopify and a connected device, so probably librespot was responsible. And I can't really reproduce it since I cannot get spotify connect back to work, but I enabled debug now and will tell when I find something.
As far as I can see RC5 is using mopify 1.6, and it prompts an update notification to 1.61.