pimusicbox/pimusicbox #451: Masalah kkarnel tp link tl-wn722n V2.0

clarksye, 2017-11-10:
i have a problem with tlwn722n v2.0 that i have, so i have installed kernel driver rtl8188EUS_linux_v4.3.0.8_13968.20150417 and rtl8188eu. I installed it in accordance with the directives of README.md, finished installing tlwn722n I can go into (monitor mode) well after that I restart my computer, but my computer fits back up and I try again to go into monitor mode is not (i say WTF! :) "), and i tried removing all drivers about tlwn722n ie: rtl8188EUS_linux_v4.3.0.8_13968.20150417, rtl8188eu and 8188eu.ko on all my computers, and i tried again to install that driver back, but only the driver rtl8188EUS_linux_v4.3.0.8_13968.20150417 which I can install because of which (rtl8188EUS {make error 2 *})
and i try to apt-get update, apt-get upgrade and apt dist-upgrade ,, it turns out my version is up to date! so that's not the case, and (linux header) is the latest one ,, i'm confused and i bucked for (ifconfig wlan0 down, iwconfig wlan0 monitor mode) JREEENG! it turned out to be and I was happy, and I restart my computer again, and I try to monitor mode again and WOW it works! ,, but that's the problem now TP LINK TLWN722n v2.0 I can not stop or switch from WTF monitor mode !,, so I want to ask if it's a problem or not? , THIS THE IMAGE I TAKE
# Helpmeforthis
[image]
[image]
[image]
[image]
[image]
[image]
[image]

thankyou ;)

clarksye, 2017-11-10:
sorry my english is not good ,, that i translated in google translation

clarksye, 2017-11-10:
makud above is i can not stop from monitor mode ,, but i can still use internet like always, i want to ask if it is problem or not and is this monitor mode working fine or not?

It's all because I deleted all rtl8188eu drivers

kingosticks, 2017-11-10:
This has nothing to do with pimusicbox. Please find a different place to ask your question.

closed 2017-11-10 by kingosticks