pimusicbox/pimusicbox #453: librespot / spotify connect failing on FiiO USB DAC

arthurlutz, 2017-11-16:
I know librespot support is beta, so I don't expect a fix, but putting this here in case some else finds a way to fix this and can help me out. 

```
RUST_BACKTRACE=1 /opt/librespot/librespot -n MusicBox -u snip -p snip  --onstart /usr/bin/mpc stop -v -b 320  --device plughw:1
[snip]
INFO:librespot::util: Exit status: exit code: 0
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /checkout/src/libcore/option.rs:329
stack backtrace:
   0: std::sys::imp::backtrace::tracing::imp::unwind_backtrace
TRACE:librespot::audio_file: chunk 6 / 92 complete
TRACE:librespot::audio_file: requesting chunk 7
             at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1: std::sys_common::backtrace::_print
             at /checkout/src/libstd/sys_common/backtrace.rs:71
   2: std::panicking::default_hook::{{closure}}
             at /checkout/src/libstd/sys_common/backtrace.rs:60
             at /checkout/src/libstd/panicking.rs:355
   3: std::panicking::default_hook
             at /checkout/src/libstd/panicking.rs:371
   4: std::panicking::rust_panic_with_hook
             at /checkout/src/libstd/panicking.rs:549
   5: std::panicking::begin_panic
             at /checkout/src/libstd/panicking.rs:511
   6: std::panicking::begin_panic_fmt
             at /checkout/src/libstd/panicking.rs:495
   7: rust_begin_unwind
             at /checkout/src/libstd/panicking.rs:471
   8: core::panicking::panic_fmt
TRACE:librespot::audio_file: chunk 7 / 92 complete
TRACE:librespot::audio_file: requesting chunk 8
             at /checkout/src/libcore/panicking.rs:69
   9: core::panicking::panic
             at /checkout/src/libcore/panicking.rs:49
  10: <librespot::audio_backend::alsa::AlsaSink as librespot::audio_backend::Sink>::start
  11: librespot::player::PlayerInternal::run
  12: std::panicking::try::do_call
  13: __rust_maybe_catch_panic
             at /checkout/src/libpanic_unwind/lib.rs:98
  14: <F as alloc::boxed::FnBox<A>>::call_box
  15: std::sys::imp::thread::Thread::new::thread_start
             at /checkout/src/liballoc/boxed.rs:650
             at /checkout/src/libstd/sys_common/thread.rs:21
             at /checkout/src/libstd/sys/unix/thread.rs:84
DEBUG:librespot::player: drop Player[0]
```

When I switch to ```--device plughw:0``` it works on the default pi soundcard (which I guess I could connect up to my amp on different channel and use that awaiting a fix). (```temp fix in /opt/musicbox/startup.sh for now```).

I thinks it's something to do with librespot trying to talk S16_LE and my DAC only supporting (or only announcing) SE32_LE see https://github.com/plietar/librespot/issues/162

```
root@MusicBox:/cf# cat /proc/asound/DAC/pcm0p/sub0/hw_params 
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 441
buffer_size: 8820
root@MusicBox:/cf# cat /proc/asound/DAC/stream0 
SmartAction FiiO USB Audio Class 2.0 DAC at usb-3f980000.usb-1.2, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us
```

maybe some sort of fix lies in https://github.com/rickyzhang82/bit_perfect but I don't get it (yet).

kingosticks, 2017-11-16:
Are you sure it's the same issue? It doesn't have the same sample format error message and alsa's plughw should solve that anyway (as per the linked issue). The log looks more like it handled a few chunks ok and then something went wrong on the 8th one. My rust experience is in its infantcy (and only due to what I've tried to understand in librespot) so maybe I'm wrong. Unfortunately it's no longer actively maintained so we might have to work this out between us.

light2yellow, 2017-12-22:
Try a recent kernel (4.14.x), looks like there had been a bug which was fixed. Pulseaudio went into D state every time as soon as I plugged in the DAC while working with ALSA just fine, now it all works as supposed.