pimusicbox/pimusicbox #454: YouTube play doesn't work

Ridicolos, 2017-12-04:
If I add a YouTube song to my queue it doesn't play it.
When I then go to the "Now Playing" menu i just see "(sync)" and the thumbnail of the song.
After several minutes of waiting nothing happens.

Any ideas? :c
[image]

kingosticks, 2017-12-04:
Some songs are using a format that's incompatible with pimusicbox. Could you provide the url for the exact song?

Ridicolos, 2017-12-04:
Sure. https://www.youtube.com/watch?v=Etif6sR-Bsc

ivanmayes, 2017-12-09:
Have this issue too.  Works on my local OSX machine with mopidy installed through Homebrew.  Tried 15 videos with the same results on my pi and couldn't get one to play.