pimusicbox/pimusicbox #455: [mpd client] playlist tracks never finish loading

porjo, 2017-12-13:
Using an MPD client, when I select a Spotify playlist the tracks always show as 'Loading' no matter how long I leave it. 

[image]

The web client does not have this problem.

I've tried 2 different MPD clients: MPDroid and Gnome Music Player - both have the same issue.

(using 0.7.0rc5)

kingosticks, 2017-12-13:
Can you please enable the debug log in settings.ini, reboot, reproduce the issue and then provide the log. You'll find it at /tmp/mopidy-debug.log

porjo, 2017-12-13:
See debug log attached here: [mopidy-debug.log](https://github.com/pimusicbox/pimusicbox/files/1557262/mopidy-debug.log)