pimusicbox/pimusicbox #456: build pimusicbox using linuxkit

arthurlutz, 2017-12-18:
I'm a big fan of debian and would love to have pimusicbox running on a debian where I can SSH in and add some stuff and upgrade some packages... but... I've come across linuxkit, which seems interesting for pimusicbox

https://github.com/linuxkit/linuxkit  A toolkit for building secure, portable and lean operating systems for containers

Although this goes in the other direction of having a "hackable" pimusicbox based on a pure debian, I think it could still be an interesting path to build the distro / images for pimusicbox. 

I was inspired by a talk (in french) about linuxkit https://www.hooktube.com/watch?v=CWTJdve0g6U which explains quite well how the "embedded, task specific, linux distros" could benefit from this. I'm sure someone can find a decent intro on youtube if the project page is not explicit enough https://hooktube.com/results?search_query=linuxkit 

Just putting this idea out there, in case someone else is tempted by this path and wants to share ideas and trying it out.

adamcik, 2019-07-21:
Thanks for the suggestion, closing in favor of #6 which already tracks a similar idea.

adamcik, 2019-07-21:
Or, would close if I could :-)

closed 2019-07-21 by kingosticks