pimusicbox/pimusicbox #457: Config Files Differ

smbenson0712, 2017-12-22:
I'm new to PMB, but got it initially working last night.  I started over as for some reason I can no longer get TuneIn to show any stations.  So, on a fresh install, I booted first time with no edits to settings.ini.  After boot, editted setting via web interface to add ssh and turn on a shairplay-sync and upnp.  Mopidy server shows up fine on MPod, but will hang on getting songs and not connect.  I check the settings.ini and mopidy.conf and they are identical for both http and mpd.  host is 0.0.0.0 and zeroconf = "" (default settings).  When I execute mopidy config, the http and mpd sections have host = 127.0.0.1. and several other settings in the mpd section that are not in either settings.ini or mopidy.conf.  I know that using loopback host is causing MPod not to connect, but where would I find the config file that is currently being used my mopidy and change it to be inline with settings.ini/mopdiy.conf.

smbenson0712, 2017-12-22:
Found the default config in /root/.config/mopidy/mopidy.conf.  I am assuming that if you want to change any of the values (uncomment them) and make a value change (i.e. host = 0.0.0.0) that these become the new defaults.  Shouldn't changes to values in settings.ini be applied after the default values if the keys are the same?

kingosticks, 2017-12-22:
When mopidy runs as a system service, as it is designed to do in pimusicbox, it uses the config file in /etc/mopidy/mopidy.conf. As it happens, in pimusicbox this file is actually a symlink to /boot/config/settings.ini. That root config file is used when you run mopidy as the root user I.e. not a system service. The default are actually elsewhere and really don't need to be touched ever. I have not tried mpod in a long time but I do remember it being a poor experience.

smbenson0712, 2017-12-22:
Thanks for the quick response.  If mopidy is running as a service, is there a special way to check it's loaded config that is different from checking mopidy config?  I can't run mopidyctl config as mopidyctl is not found.  It seems that changes made to /etc/mopidy/mopidy.conf are not being made when the service is restarted.  I can tell because the zeroconf name is staying at the default even if I change it in settings.ini or conf file directly.

kingosticks, 2017-12-22:
You don't really want to be changing zeroconf in there, I think the comment should specify that it's set like that to disable it. That's because we use the system zeroconf settings for everything rather than having various services implement it themselves. I'd have to check where that zeroconf file lives I don't remember off the top of my head.

I also don't recall what the appropriate alternative to mopidyctl is in pimusicbox v0.7. I think it might be as simple as 'service mopidy run config'.

smbenson0712, 2017-12-22:
"service mopidy run config" works.  I'm not worried about the name, but I was changing it because it shows in MPoD even if I don't connect to it during this testing.  If I change the root's hidden config host info to 0.0.0.0, the service config changes accordingly and connection to MPoD works fine and operation is normal.  Now I just have to figure out why I get no TuneIn stations being displayed.  Last night I was getting everything listed just fine.  Any ideas on troubleshooting TuneIn?  In the service config it is enabled and timeout specified.  All I get is Trending with nothing in it when I browse there in MPoD or web client.

smbenson0712, 2017-12-23:
Came home 3 hrs later tonight and TuneIn is showing everything fine.  Not sure what the issue was, but it is working fine.  Thanks for the info on configs - 2 more Pi's ordered to replace my old asus routers running mpd.

smbenson0712, 2017-12-27:
There are still some settings that are not mapping through settings.ini to the running mopidy config.  In settings.ini I have turned off everything except tunein, but after save and reboot, when you hit browse in the webclient,  files, spotify browse, tunein, itunes-podcasts are still there.  When I run service mopidy run config, it shows podcast-itunes true (podcast shows false), spotify_tunigo true (spotify and spotify_web show false), and file true (local shows false)  It looks like there is a conflict in labels between settings.ini and mopify.conf.  I hope the way I'm explaining this makes sense.  I would try to disable these manually, but I don't want to mess anything up.

smbenson0712, 2017-12-29:
You can close this - I know what is going on, but use it for your info if you want settings.ini to configure all the options in the default config and clean up things so the ui can control all the items displayed in browse.  Thanks for the quick responses.

closed 2017-12-29 by kingosticks