pimusicbox/pimusicbox #460: Now Playing screen over HDMI

spacecakes, 2018-01-05:
I'm using Pi MusicBox via HDMI to my AV system mostly to play songs over Spotify Connect. It would be cool to have a 'Now Playing' screen directly on the TV instead of staring at the CLI.

MicahJMungal, 2018-01-20:
You should check out volumio.  Similar to pimusicbox but there is a plugin from the developers to add touch screen functionality.

kingosticks, 2018-01-20:
@spacecakes this something I'd love to have in pimusicbox. We could hook it up to simple commands from the TV remote control with libcec too. Ideally I'd like to avoid bundling x11 in the image, it seems over the top. Maybe some other simpler window manager would be good, or perhaps an interface written in pygame (or kivy?) as I think they don't need a window manager, but I'm not sure. Failing that, if we did go for x11 then you launch a browser in kiosk mode to show a webclient.

MelBourbon, 2018-05-15:
I would like to see nothing. How can I disable CEC? Because I have my PI also connected per HDMI and at every start of my TV the source of the PI is chosen because the PI sents a contstant CEC signal. How can I disable this?

kingosticks, 2018-05-15:
Add `hdmi_ignore_cec=1` to your  config.txt

spacecakes, 2021-04-15:
@kingosticks oh so you like this feature. :-)

kingosticks, 2021-04-15:
Yes!

> this something I'd love to have in pimusicbox