pimusicbox/pimusicbox #462: Add Second Instance of Mopidy & Second Sound Card

smbenson0712, 2018-01-24:
Do you have a recipe for adding a second mopidy service that would drive a second usb sound card?  I have a multi-source amp/multi-room system that I currently have a roku soundbridge and a pi sending output as 2 sources.  My roku is dying, so instead of having 2 pi's serve two sources, I would like to just add a second sound card and reconfigure the pi musicbox to handle both source outputs.  I would also like to figure out how to have shairport-sync have two different instances and names so it sent to the correct source.

Wondering if it would also be possible to use both the wifi and wired networks of the pi3 to assign separate IP address for each instance?  If not, I could also specify different ports for web/app control.

smbenson0712, 2018-01-25:
I figured out how to get two instances of both mopidy and shairport-sync running on two different usb sound cards after a night of searching / trial and error.  I used separate ports for the http section and haven't tried both wifi/wired interfaces yet.  

The only problem I have now is getting mpc to stop the second instance of mopidy so shairport-sync can start on the second sound card.  If mopidy2 is not playing, shairport-sync will play.  I will post on Mopdiy Discourse.

closed 2018-01-25 by smbenson0712

kingosticks, 2018-01-25:
Good idea, I don't have a recipe for that so best ask there. 

I would say that since mpc uses the mpd extension for Mopidy you'll need to setup different ports for that too. And then have shairport's before-play hook thing configured to call `mpc -p <the port> as applicable. Good luck.

smbenson0712, 2018-01-25:
Figured it out about an hour ago.  Thanks again for helping out.  

Is it weird that every time I mess with one of my 3 music boxes, my tune-in radio stations go away?  They have been working fine to the last couple months and today they are gone again??

kingosticks, 2018-01-25:
What do you mean by go away? Nothing appears in any of the browse categories? No search results? Saved streams (in a m3u playlist/musicbox favourites) don't play?

smbenson0712, 2018-01-25:
When I browse to TuneIn there is only a Trending Folder and that is empty.  No search results.  Even the streams that I have saved in favorites won't play.  I can play them on my computer, so Tune In is up and I can access via web browser.

kingosticks, 2018-01-25:
You can get in a state, generally with badly behaving mpd clients, where the extension tries to do too many requests to the tunein API and they get rate limited (temporarily blocked). The extension is a bit simple and doesn't handle this well. But if you look in the log file you should see loads of error messages if this is what's happened. Does this issue persist between restarts (of mopidy or the system)? Could you check the Mopidy log at /var/log/mopidy/mopidy.log. If nothing in there, could you try enabling the debug log and see what comes out then?

smbenson0712, 2018-01-25:
Getting a warning  - problem looking up http://149.56.240.65:20056 - so you are probably right.  I suppose restarting a bunch while troubleshooting configs would cause too many requests.  

Getting another bunch of warnings - Element doesn't implement handling of this stream. Please file a bug.  No sure what that would be, Everything I have played on this unit has played fine (internet streams from a playlist).

kingosticks, 2018-01-25:
You have to do quite a lot of requests to hit the rate limiting and the request would be to tunein.com rather than specific stream addresses (like that looks to be). Hmm.

That element handing warning is indeed a gstreamer red herring, just have to ignore that one I'm afraid.

smbenson0712, 2018-01-25:
On this unit I don't even have any streams from tune in saved in the playlist.  The other ones do, but I haven't even been using them.  I had a similar problem when I first started using musicbox a couple of months ago.  I was fiddling alot with configs/setting trying to get things working and tune in would do the same thing.  Left home for 4-5 hours and it worked fine after I got home.  Is there anything else I can do on my end to see if sending a large number of requests that would cause this?

kingosticks, 2018-01-25:
You could turn on the debug log, restart mopidy and then browse the tunein section. You should then be able to find the http request to api.tunein.com (or something like that) that is failing. If it's failing due to rate limiting there should be an error message to that effect. Failing that, if you could share the log then I can try to reproduce the issue tomorrow.

kingosticks, 2018-01-25:
If it is rate liming then note its per ip address. So hitting their service from other musicboxes / phones / PCs all adds up.

smbenson0712, 2018-01-25:
I restarted and logged in via webclient and browsed to tune in folder and trending.  Also did the same with my iphone using MPoD.  Here is the debug log.
[mopidy-debug.log](https://github.com/pimusicbox/pimusicbox/files/1665915/mopidy-debug.log)

kingosticks, 2018-01-25:
Eek, perhaps they have changed the API.
```
urllib3.connectionpool: http://opml.radiotime.com:80 "GET /Browse.ashx?render=json&c= HTTP/1.1" 403 1233
2017-09-20 02:30:36,832 INFO [2075:TuneInBackend-6] mopidy_tunein.tunein: TuneIn API request for Browse.ashx failed: 403 Client Error: Forbidden for url: http://opml.radiotime.com/Browse.ashx?render=json&c=
```

kingosticks, 2018-01-25:
Hmm, no, works for me. But they don't seem. To like you. I would hope it's just a rate thing that will expire in a few hours. I will look to add something more intelligent to tunein. The problem is that the whole thing is reverse engineered (in the most trivial sense) so I don't know what the rates are.

smbenson0712, 2018-01-25:
It seems to not like me when I start messing around with configs etc.  I never even look at it during the process, but when I finally got it working I checked and it was gone again.  Is there possibly something that sends out requests on service restarts/reboots that would spam them even if you don't actually browse there because the mopidy-tunenin module is enabled?  Like I said it has happened to me when I was messing with config files before trying to get things working on other units as well.  Then boom it's back working.  Thanks again for looking at this.  

I will try to put together a how to for multi-instance mopidy/shairport-sync soon.  I will post it on mopidy discourse as it seems several people had questions there.

Would there be anyway to build an aggregate web interface for multiple units/instances?

kingosticks, 2018-01-25:
No requests are made until you try and browse the tunein section or submit a search or if you load a playlist that includes  a tunein entry. And that really should be fine, the rate limit kicks in when you send literally hundreds of requests back-to-back. I've seen it happen with badly behaved mpd clients that try and locally cache the entire database (which is a big no). I don't remember which mpd clients do this.

An interface that includes a way to switch between multiple musicbox interface is something I want to include. I was generally thinking they would be on separate Pis as that's the more common case but I guess that's not important. Sadly the idea hasn't progressed yet.

smbenson0712, 2018-01-26:
I would love to replace MPod, but to my knowledge none of the iOS clients allow for multiple servers like MPod.  Until I find an alternative with multi-server/instance option, I'm sticking with iOS 10.  MPod is not perfect, but it works for me.  Thanks for all your work on pi Musicbox.  It's a great distro.

smbenson0712, 2018-01-26:
Tune In is working fine again.  It has to be MPod spamming TuneIn with requests when I setup a new musicbox.