pimusicbox/pimusicbox #463: Roadmap for multi-room?

heylookltsme, 2018-02-05:
I'm wondering if you guys have a roadmap for the multi-room audio feature.  I'm particularly interested in the following: 
- a single web UI for controlling playback in all rooms
- streaming the same audio feed to multiple rooms in sync

(aka a sonos-like experience) 

Any plans for adding either of those things in the near future? Thanks!

ekennedy80, 2018-02-06:
I am also very interested in this feature

kingosticks, 2018-02-06:
I'd love to add this and in that respect its on the road map, pretty much exactly as you describe. Some people have got the 2nd point going by installing snapcast, if you search on http://discuss.mopidy.com there should be something on there (I can't find the particular example I'm thinking of right now).

A UI for showing/selecting other musicbox clients integrated into the current webclient would be a big help. Would be great if anyone wanted to start with that. Ideally some self-contained code to listen for other clients (via mdns) and expose the data for a Web page to display (musicbox-webclient, at least initially).

Funkwarrior, 2018-02-22:
What about https://github.com/badaix/snapcast ?

kingosticks, 2018-02-22:
Yes, probably snapcast as I mentioned above.

Funkwarrior, 2018-02-22:
oh sorry, missed that part.. sorry for my useless comment.
Keep on with your wonderful work!

kingosticks, 2018-02-22:
I'm not sure what to do about the UI for it all though. Love to hear any thoughts people have.

BLX-S, 2018-05-24:
@kingosticks Snapcast has an android app which communicates with the snapserver. You could try and port over the controls used there into the musicbox webui. I personally really like what they did at: https://github.com/nathanborror/Play