pimusicbox/pimusicbox #464: Issues when Spotify password contains a hash

andrewguy, 2018-02-08:
There is some strange behaviour going on when the Spotify password in `settings.ini` contains the `#` symbol. For arguments sake, let's say my old password was `old#pw`, and the new password was `newpassword`. 

I initially set the password in the `settings.ini` file:

`password = old#pw`

I was able to connect to Spotify at this point in time.

I then changed the password through the web interface, as I'd noticed that the password was only displaying as 3 characters long (i.e. it was reading everything after the `#` as a comment). I changed my spotify password to remove the `#` - now `newpassword`. I updated this via the web interface, but couldn't get the Spotify to connect anymore.

After some troubleshooting, I had another look at the `settings.ini` file. The password line was now:

`password = newpassword#pw`

It seems to have kept everything after the `#` from the old password, but replaced the preceding text with the new password.

**TL;DR**

It seems that mopidy is reading the Spotify password line in `settings.ini` correctly (even if it contains `#`), but changing the password via the web interface introduces errors as it treats a `#` in the password as the start of a comment, and doesn't remove characters after the `#` when updating a password.

kingosticks, 2018-02-08:
Thanks for the detailed report. It does indeed sound like a problem with websettings, I guess with the underlying ini parser we're using. The current parser is a pita in a few ways, I was thinking it'd be better just to borrow Mopidy's which is much better. It's not supposed to be (ab)used by other code as it is but maybe it'd be worth pulling it out of Mopidy as a standalone library.

andrewguy, 2018-02-08:
Yep, with a bit more digging it looks like it's the `configobj` parser that's causing the issue.

It's actually not an issue if you enclose your password in quotes, so maybe this just needs to be mentioned in the docs or setup instructions.

Alternatively, it would be possible to pull out the inline comment from the password line, and append to the current password. Something like:

```
from configobj import ConfigObj, ConfigObjError

iniconfig = ConfigObj('settings.ini')

current_password = iniconfig['spotify']['password'] + iniconfig['spotify'].inline_comments['password']
# Remove stored inline comments so we don't write them back to file when the password is changed
iniconfig['spotify'].inline_comments['password'] = None

iniconfig.write()
```

I'll need to have a longer look at the codebase to figure out if this is reasonable. 

I guess there would then be issues with this approach if the password _is_ enclosed in quotes, and someone wants an inline comment.