pimusicbox/pimusicbox #465: Logarithmic volume scale for Spotify Connect

flyser, 2018-02-10:
librespot recently got support for logarithmic volume scaling (see https://github.com/librespot-org/librespot/pull/10 ). It would be great if pimusicbox would support it as well :)

kingosticks, 2018-02-10:
> librespot-org/librespot#10

Sure, there's also support for repeat and shuffle. Have been playing with
the latest build for rc6 (aka v0.7.0)!

kingosticks, 2018-03-22:
Oops, I forgot to enable this in rc6. rc7 is coming imminently to fix the mounting issues so I will sort this too.

kingosticks, 2018-03-22:
And oops again I forgot it's enabled by default. That's twice I've made that mistake in as many weeks! So good news is that this is already available in https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC6.

closed 2018-03-22 by kingosticks