pimusicbox/pimusicbox #466: Remove subsonic support from the feature list

heylookltsme, 2018-02-19:
I just installed the pi musicbox image and was all pumped to hook it up to my subsonic library, but it wasn't working out of the box. I did some searching and was bummed to find these [two](https://github.com/pimusicbox/pimusicbox/issues/226) [issues](https://github.com/pimusicbox/pimusicbox/issues/349) which seem to indicate that pi musicbox doesn't actually have support for subsonic anymore. 😿 

I propose that subsonic be removed from the feature list in the documentation and website so other poor subsonic users won't get all sad. 

On the other hand, if there are any known workarounds, I am all ears! Thanks 👍

kingosticks, 2018-02-19:
Sorry about that. I hadn't planned to be on v0.7 for long (and thus unable to use mopidy-subidy) but things worked out a little differently. 

I will remove it from the feature list with the intent of adding it back in the future.

closed 2018-02-19 by kingosticks

heylookltsme, 2018-02-19:
Thanks @kingosticks! 👍

heylookltsme, 2018-02-19:
Also, just for posterity in case someone else comes along hoping for subsonic support, I was able to do without it pretty easily. My subsonic server is running on another raspberry pi that has a hard drive full of my music mounted on it. I just served up that hard drive as a samba share and used pi musicbox's network drive settings to access it. Et viola. ✨

commit e6f2d68b (2018-02-19): Remove subsonic from current feature list (Fixes #466).
commit 58435d2b (2018-02-19): Remove subsonic config (Fixes #466).