pimusicbox/pimusicbox #467: 404 Error on Pi MusicBox Web Client

PiFi77, 2018-03-01:
After flashing Pi MusicBox to an SD card and powering up, I found that I cannot access the web client via the assigned Pi IP (192.168.1.8). All I get is a 404 error. It using ethernet rather than wifi and my gateway is showing it active on my network. Not sure how to proceed on this as I've tried everything I can think of. As of yet I've not managed to get to the MusicBox web settings page even once since starting this project.

Any help would be greatly appreciated.

kingosticks, 2018-03-22:
There is a new release at https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC6 but there's nothing specific in there to address this as I cannot reproduce. 

Generally, if you can't access the webclient it's because Mopidy isn't running. You can plug in a monitor or login via ssh and view the log file at /var/log/musicbox_startup.log to see why.

xTipo, 2020-01-27:
Hello, i have same problem here. Just fresh installation of MusicBox adjusted config.ini.
All seems fine,  but webinterfaace did not work.
SHH connection is OK.
Checking booting Log no issue found.
When try to reach the machine by single ip ssems he try to reach some intex.html with 404error

Can you kindly advice what to check.

kingosticks, 2020-01-27:
Can you provide a copy of the log file here?

xTipo, 2020-02-01:
Hello, attached is the log file for your refer, please check, thanks in advance 

[image]

xTipo, 2020-02-05:
Hello, can somebody kindly look what happen? Log file above, thanks in advance.

elpolak0, 2020-06-01:
hello,
i have the same problem with the latest version. Is there already a solution?

michaelsalmons, 2020-10-09:
I have exactly the same issue, except my log file shows Mopidy is not starting. The only things I have enabled in settings.ini are Spotify and Spotify connect. how do I get Mopidy to run so I can access the web interface?

kingosticks, 2020-10-12:
Please provide your musicbox startup log. You can manually start it by running `/etc/init.d/mopidy start`

michaelsalmons, 2020-10-12:
Sure... thanks. I ran the start Mopidy command and here is the log.
`************************
Initializing MusicBox...
************************
Setting sound configuration...
Probing sound devices:
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
* Found internal device: card0
* Found internal device: card0
Selected card=0  (i2s=  output=analog  usb=  intc=0)
Using audio card0 (analog)
MusicBox name is musicbox
ifup: interface wlan0 already configured
Running /etc/init.d/networking restart is deprecated because it may not re-enable some interfaces ... (warning).
Reconfiguring network interfaces...RTNETLINK answers: No such process
Internet Systems Consortium DHCP Client 4.2.2
Copyright 2004-2011 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/b8:27:eb:5a:76:c8
Sending on   LPF/eth0/b8:27:eb:5a:76:c8
Sending on   Socket/fallback
done.
Stopping Samba daemons: nmbd smbd.
Starting Samba daemons: nmbd smbd.
Waiting for network (1 of 4)...

Running /etc/init.d/networking restart is deprecated because it may not re-enable some interfaces ... (warning).
Reconfiguring network interfaces...done.
Starting Dropbear SSH server: dropbear.
Starting Spotify Connect : librespot.
Starting daemon monitor: monit.
***********************************************************************************
My IP address is 192.168.0.11
Connect to me in your browser at http://musicbox.local or http://192.168.0.11
***********************************************************************************
`

kingosticks, 2020-10-12:
Weird, any clues in /var/log/mopidy/mopidy.log as to why it didn't start?

michaelsalmons, 2020-10-12:
mopidy.log consists exclusively of this, repeated over and over:

 WARNING  /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
  utils.CryptographyDeprecationWarning

michaelsalmons, 2020-10-12:
It occurred to me to update OpenSSL, but I can't do it through apt. Apt-get update yields errors:

 Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/wheezy/rpi/source/Sources  404  Not Found [IP: 93.93.128.193 80]

I assume because Pi MusicBox is a few versions behind.

kingosticks, 2020-10-12:
Yes, but you should not need to update it to start Mopidy. it should print out a lot more information than that e.g. the version.