pimusicbox/pimusicbox #468: Device no longer visible via Spotify Connect

grawlybean, 2018-03-03:
Hello.

My Pi Musicbox has been working beautifully with Spotify Connect until recently when it disappeared from the device list on IOS and Android. Reboots do not solve things.
I am using the latest RC5 release.

Any suggestions? I am wandering if a recent update to the Spotify App has messed things up?

Thanks

moonman91, 2018-03-03:
Hello,

I have been using PiMusicBox for a few days now and I have the same issue. The only way I found to make it work again is:
Disable Spotify connect
Reboot
Enable Spotify connect
Reboot

This is not very handy but it does work for me.

krzysiacz, 2018-03-03:
I am sorry. The disable/reboot sequence did not help here.
My RaspberryPi/PiMusic based player appeared just for a few seconds in the device list on my iOS-device. During that time I was unable to switch to the Raspberry. Other commercial devices on my network work fine. 
All I can do with my RaspberryPi is play spotify-playlists. 
This switches the output on other devices off (which is what i expect).
On the other hand I must confess I do not feel very comfortable with the PiMusic RC5 download.
As far as I understand, the network settings in the installed raspbian do not follow the current practice.
I do not see systemctl. And the call /etc/init.d/network issues a warning.

grawlybean, 2018-03-03:
Thanks but the reboot does not work for me either. Disappointing as pimusicbox is brilliant and I love being able to control things via the web client or Spotify Connect. I have been using versions of this excellent software for a few years now.
I’ll keep looking but might have to try Raspotify as an alternative. I suspect Spotify may have changed the API and my recent App updates are not now compatible?

kingosticks, 2018-03-03:
I think there's been some talk on the librespot mailing list about some
incompatibilities with the latest Spotify client. I'm out of the country
right now so I can't test this myself but I've got a new version to release
which has the latest fixed librespot. I'll get it uploaded when I'm back
next week.

This kind of thing is going to happen all the time since librespot is
entirely reverse engineered and Spotify frequently break things.

krzysiacz, 2018-03-03:
Yes, I guess we are all aware of the status. 
Thanks for your work.

closed 2018-03-03 by grawlybean

kingosticks, 2018-03-03:
OK, and remember there is not a lot stopping you from taking any more resent librespot build and overwriting the binary in musicbox.

grawlybean, 2018-03-03:
Ok thanks. I’ll give that a go. And thanks again for this excellent software.

krzysiacz, 2018-03-04:
As I see it: one has to compile the  librespot. 
The dirty fix: I extracted the librespot from the raspotify raspbin package.
In order to use this lib in PiMusic I had to comment the --onstart parameter in /opt/musicbox/startup.sh.
After reboot I can switch as usual to my rasppberry in Spotify Connect.

moonman91, 2018-03-04:
@krzysiacz Could you give details on how you do this please? I don't have a big knowledge of all of this.

krzysiacz, 2018-03-04:
First: this is a quick and dirty fix. 
I am not sure if it is any good in the long run.
My raspberry still works though.
The fix:
There are two files you have to change in the PiMusic system.
After you have enabled ssh you can copy the files to the right destination using scp command on Linux/macos.
(I am sure on windows putty is also able to copy files.) 
The two files are librespot and startup.sh.
The destiantion path on the musicbox is:
/opt/librespot/librespot
/opt/musicbox/startup.sh
Both files have to owned by root and have the right permission.

So the copy commands would be:
scp librespot root@<name of your musicbox>:/opt/librespot/librespot
scp startup.sh root@<name of your musicbox>:/opt/musicbox/startup.sh
followed by a 
ssh root@<name of your musicbox>
and on the musicbox:
chmod 755 /opt/librespot/librespot
chmod 755 /opt/musicbox/startup.sh

Reboot and have a look at 
/var/log/musicbox_startup.log
on the musicbox for possible errors.

Now, if you still are willing to change your system: where can I put the files?

moonman91, 2018-03-04:
Thank you for the explanation. Could you use wetranfer (https://wetransfer.com/) for the files?

krzysiacz, 2018-03-04:
https://we.tl/KVLligxMEW

grawlybean, 2018-03-04:
Very many thanks @krzysiacz
Got my system back up again

AlbertMN, 2019-03-19:
For those who (like I) find this issue, a good fix can be [found here](https://github.com/pimusicbox/pimusicbox/issues/480#issuecomment-392972884) (solved by copy-pasting a few lines). Still does the trick roughly a year after the response was made :)