pimusicbox/pimusicbox #469: Musicbox stopped showing up as a device in the Android app

kovacszo, 2018-03-03:
I have been using the Musicbox for months, but today when I wanted to use it I realized that it is not discovered by the Spotify app. Every time I restart the pi, it shows up in the app just as before, but after a few seconds it disappears. The web UI of the MusixBox works well, it can connect to the service and plays the tracks I start there. 
Probably the latest version of the Spotify app (8.4.42.722) brought some change.

Any help is appreciated.

ramonhollands, 2018-03-12:
I've got the same issue. Also on the spotify app on my macbook air.

rvliege, 2018-03-15:
Same behaviour here. After fresh install it works for some minutes. Looks like ID get banned or something like that.

kingosticks, 2018-03-22:
https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC6 has the latest version of librespot which fixes the recent issues.

rvliege, 2018-03-23:
RC6 has the same behavior after about 10 hours of being discovered by desktop and mobile (Android) client I lost it again. No change in desktop or mobile client.

kingosticks, 2018-03-23:
You would need to check /var/log/librespot.log for a stack trace. I imagine librespot errored and unfortunately when it does this it doesn't actually exit, it just gets stuck in a useless state. The librespot project is working towards fixing this kind of thing but it's still a work in progress. I have tried to force it to restart in rc7 (not yet released) but I don't know hoe well that is going to work. I could add a button that allows you to restart the service from the web ui.

rvliege, 2018-03-23:
I forgot to mention that restart of the device doesn't fix the problem. At least that's what I'm experiencing.
I will check the log for more info.

kingosticks, 2018-03-23:
On Android I have found I sometimes need to restart the spotify client too.

rvliege, 2018-03-23:
Did a restart of the MB device and client every time without success. Desktop is win32 application and Android app. After boot the player is available for connect shortly. Connecting is impossible. So behavior is same as @kovacszo describes but now in RC6.

My logging:
```
INFO:librespot: librespot 9d9c311 (2018-03-11). Built on 2018-03-13. Build ID: BECir1hc
INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-b-n1bs.ap.spotify.com:4070"
thread 'main' panicked at 'Authentication failed with reason: BadCredentials', core/src/connection/mod.rs:85:21
stack backtrace:
   0:   0x89263f - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h280c621ec0336843
                       at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1:   0x89c89b - std::sys_common::backtrace::print::h5bb167d48668dac9
                       at /checkout/src/libstd/sys_common/backtrace.rs:68
                       at /checkout/src/libstd/sys_common/backtrace.rs:57
   2:   0x88aec7 - std::panicking::default_hook::{{closure}}::hccf3b2e1cc48596f
                       at /checkout/src/libstd/panicking.rs:381
   3:   0x88a9a3 - std::panicking::default_hook::h338b1189515ccc69
                       at /checkout/src/libstd/panicking.rs:397
   4:   0x88b373 - std::panicking::rust_panic_with_hook::h5d950fb4a856fba5
                       at /checkout/src/libstd/panicking.rs:577
   5:   0x88b217 - std::panicking::begin_panic::h7a5157aecf168062
                       at /checkout/src/libstd/panicking.rs:538
   6:   0x88b143 - std::panicking::begin_panic_fmt::hc44606ee40e9a98e
                       at /checkout/src/libstd/panicking.rs:522
   7:   0x5a62df - <futures::future::chain::Chain<A, B, C>>::poll::h3fcdaa58a6d3face
   8:   0x5ee477 - <futures::future::and_then::AndThen<A, B, F> as futures::future::Future>::poll::habc36517dd5700b9
   9:   0x5aa24f - <futures::future::chain::Chain<A, B, C>>::poll::h6d0ada6a6e479f45
  10:   0x5f6aab - <futures::future::map::Map<A, F> as futures::future::Future>::poll::ha18fc3b66722a846
  11:   0x475c37 - <librespot::Main as futures::future::Future>::poll::ha34952329266c131
  12:   0x46e61f - futures::task_impl::std::set::haa73a9ee0a468b94
  13:   0x486f8b - tokio_core::reactor::Core::run::h8b40cb76d6ebd0af
  14:   0x477aa3 - librespot::main::hf84f741837bd497f
  15:   0x487a13 - std::rt::lang_start::{{closure}}::h369662dc239e3883
  16:   0x88b01b - std::panicking::try::do_call::h9cb586bd57bb9d3d
                       at /checkout/src/libstd/rt.rs:59
                       at /checkout/src/libstd/panicking.rs:480
  17:   0x8b1c33 - __rust_maybe_catch_panic
                       at /checkout/src/libpanic_unwind/lib.rs:101
  18:   0x8922cb - std::rt::lang_start_internal::h104139376fdb9ead
                       at /checkout/src/libstd/panicking.rs:459
                       at /checkout/src/libstd/panic.rs:365
                       at /checkout/src/libstd/rt.rs:58
  19:   0x478583 - main
  20: 0xb6dc981b - __libc_start_main
```

Does this mean the credentials of my Spotify account are wrong? These aren't necessary for the use of the connect feature?

rvliege, 2018-03-27:
Okay it took some time for me to test this. I filled in my Spotify credentials and Mopidy credentials a couple of days ago and checked desktop and Android client if the MB showed up. Unfortunately only shortly after restart. Can't connect tho. Guess this issue still persists.

kingosticks, 2018-03-27:
I've had some similar issues myself with my model 2 pi. Sometimes it connects and works fine, sometimes it crashes straight away, sometimes it works fine at first but then connecting again later and it'll crash straight away. It seems to be more reliable on my model 3 but I'm still not convinced it's solid there either. I don't think there's much I can do here about this; it's an upstream issue for the librespot folk and one they do know about.

kevinpcassidy, 2018-04-18:
I'm a bit new to this, just got musicbox up and running a month ago, and am having the same issue as posted here. Was having inconsistent issues in March, and updated to the newest image RC6 last night.  Spotify connect worked flawlessly for the evening, and today, it recognized musicbox briefly, then lost the connection. I've tried rebooting and enabling/disabling various Spotify parts in the settings. I've also noticed that it's not able to access any Spotify music when browsing on the musicbox web app. Fyi-  I'm running musicbox on a pi zero w. Is it more stable on a pi 3?

MartyTremblay, 2018-05-08:
In my case, noticed that special characters in my password was throwing the same error. If I add single quotes around my password "-p" for DAEMON_ARGS in /etc/default/librespot, this solved the issue.

kingosticks, 2018-05-08:
I don't follow, I get a "bad credentials" error if I try and single quote the username and/or password.

MartyTremblay, 2018-05-09:
Actually, I misdiagnosed the problem. Turns out that running with the password in quotes from the shell works:

`/opt/librespot/librespot -n MusicBox -u someone@email.com -p 'someP@ssword&!' -b 320 --onevent /opt/musicbox/mpc_stop.sh --device-type speaker`

However, putting single quotes in DAEMON_ARGS does not resolve the issue. Clearly the problem is with how librespot is executed.

MartyTremblay, 2018-05-09:
there might be an issue with start_stop_daemon with how it handles special characters in the args. I've found the following:

https://stackoverflow.com/a/8333333