pimusicbox/pimusicbox #472: Element doesn't implement handling of this stream. Please file a bug.

pdnellius, 2018-03-13:
Receiving the following error, referenced in #418 [(direct link here)](https://github.com/pimusicbox/pimusicbox/issues/418#issuecomment-372721834):
```
WARNING  Element doesn't implement handling of this stream. Please file a bug.
```
I am able to play test files from the RPi just fine using the built-in audio output. Anything played through PiMusicBox outputs the above error. 

I am able to search/queue just fine, appears as though playing. 

Output of `service mopidy run config`:
```
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy
max_tracklist_length = 10000

[logging]
color = true
console_format = %(levelname)-8s %(message)s
debug_format = %(levelname)-8s %(asctime)s [%(process)d:%(threadName)s] %(name)s\n  %(message)s
debug_file = /var/log/mopidy/mopidy-debug.log
config_file = /etc/mopidy/logging.conf

[audio]
mixer = software
mixer_volume = 85
output = alsasink

[proxy]
scheme = 
hostname = 
port = 
username = 
password = 

[mpd]
enabled = true
hostname = 0.0.0.0
port = 6600
password = 
max_connections = 20
connection_timeout = 60
zeroconf = 
command_blacklist = 
  listall
  listallinfo

[http]
enabled = true
hostname = 0.0.0.0
port = 6680
static_dir = /opt/musicbox/webclient
zeroconf = 

[stream]
enabled = true
protocols = 
  http
  https
  mms
  rtmp
  rtmps
  rtsp
metadata_blacklist = 
timeout = 5000

[m3u]
enabled = true
playlists_dir = /music/playlists

[softwaremixer]
enabled = true

[file]
enabled = true
media_dirs = 
  /music|Music
show_dotfiles = false
follow_symlinks = true
metadata_timeout = 1000

[local]
enabled = true
library = sqlite
media_dir = /music
data_dir = $XDG_DATA_DIR/mopidy/local
scan_timeout = 1000
scan_flush_threshold = 100
scan_follow_symlinks = true
excluded_file_extensions = 
  .directory .html
  .jpeg
  .jpg
  .log
  .nfo
  .png
  .txt
  .mkv
  .avi
  .divx
  .qt
  .htm
  .zip
  .rar
  .gz
  .pdf
  .exe
  .ini
  .mid
  .db
  .m3u
  .sfv
  .midi

[youtube]
enabled = true

[websettings]
enabled = true
musicbox = true
config_file = /boot/config/settings.ini

[tunein]
enabled = true
timeout = 5000

[subsonic]
enabled = false  ; Extension disabled by user config.

[spotify]
enabled = true
username = [omitted]
password = ********
client_id = [omitted]
client_secret = ********
bitrate = 320
timeout = 10
cache_dir =
settings_dir = $XDG_CONFIG_DIR/mopidy/spotify
toplist_countries = 
  AD
  AR
  AU
  AT
  BE
  CO
  CY
  DK
  EE
  FI
  FR
  DE
  GR
  HK
  IS
  IE
  IT
  LV
  LI
  LT
  LU
  MY
  MX
  MC
  NL
  NZ
  NO
  PT
  ES
  SG
  SE
  CH
  TW
  TR
  GB
  US
search_album_count = 20
search_artist_count = 10
search_track_count = 50

[spotify_web]
enabled = false  ; Extension disabled by user config.

[spotify_tunigo]
enabled = true
region = 
sub_genres = true
cache_time = 3600

[soundcloud]
enabled = false  ; Extension disabled by user config.

[somafm]
enabled = false  ; Extension disabled by user config.

[simple-webclient]
enabled = true

[scrobbler]
enabled = false  ; Extension disabled by user config.

[podcast]
enabled = true
browse_root = Podcasts.opml
browse_order = desc
lookup_order = asc
cache_size = 64
cache_ttl = 86400
timeout = 10

[podcast-itunes]
enabled = true
retries = 3
country = US
explicit = 
base_url = http://itunes.apple.com/
charts = audioPodcasts
search_limit = 20
timeout = 10
charts_limit = 20

[musicbox_webclient]
enabled = true
musicbox = true
websocket_host = 
websocket_port = 
on_track_click = PLAY_ALL

[mopify]
enabled = true
debug = false

[moped]
enabled = true

[mobile]
enabled = true
title = Mopidy Mobile on $hostname
ws_url = 

[local-sqlite]
enabled = true
directories = 
  Albums                  local:directory?type=album
  Artists                 local:directory?type=artist
  Composers               local:directory?type=artist&role=composer
  Genres                  local:directory?type=genre
  Performers              local:directory?type=artist&role=performer
  Release Years           local:directory?type=date&format=%25Y
  Tracks                  local:directory?type=track
  Last Week's Updates     local:directory?max-age=604800
  Last Month's Updates    local:directory?max-age=2592000
timeout = 10
use_album_mbid_uri = true
use_artist_mbid_uri = false
use_artist_sortname = false

[internetarchive]
enabled = false  ; Extension disabled by user config.

[http-kuechenradio]
enabled = true
title = Mopidy Küchenradio
default_locale = en

[gmusic]
enabled = false  ; Extension disabled by user config.

[dirble]
enabled = true
api_key = 473279e3fa0e7010cbbbb40ecc31890d46e57a2e
countries = 
  US
  NL
  DE
  NO
  SE
timeout = 5000

[audioaddict]
enabled = false  ; Extension disabled by user config.

[alsamixer]
enabled = false  ; Extension disabled by user config.
```

output of `aplay -l`:
```
root@MusicBox:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

kingosticks, 2018-03-13:
Thanks for this. One more question (sorry), what model Pi is this?

pdnellius, 2018-03-13:
@kingosticks ,

I believe it's a Pi Model 3

kingosticks, 2018-03-22:
I have tried but I've been unable to reproduce this since I posted previously, with either rc5 or rc6. Please someone reopen if they run into it again.

closed 2018-03-22 by kingosticks

dev-danny, 2018-04-23:
hi so I'm getting this error on  0.7RC6.

heres my mopidy log
`2018-04-23 20:24:50,643 - WARNING  404 GET /favicon.ico (192.168.0.46) 23.74ms
2018-04-23 20:26:06,129 - WARNING  404 GET /favicon.ico (192.168.0.46) 19.33ms
2018-04-23 20:26:11,036 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-04-23 20:36:05,085 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-04-23 20:36:48,165 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-04-23 21:30:46,120 - WARNING  404 GET /apple-touch-icon-152x152-precomposed.png (192.168.0.21) 19.62ms
2018-04-23 21:30:46,142 - WARNING  404 GET /apple-touch-icon-152x152.png (192.168.0.21) 11.91ms
2018-04-23 21:30:46,164 - WARNING  404 GET /apple-touch-icon-precomposed.png (192.168.0.21) 11.55ms
2018-04-23 21:30:46,185 - WARNING  404 GET /apple-touch-icon.png (192.168.0.21) 11.59ms
2018-04-23 21:30:46,213 - WARNING  404 GET /favicon.ico (192.168.0.21) 11.69ms
2018-04-23 21:30:46,268 - WARNING  404 GET /apple-touch-icon-152x152-precomposed.png (192.168.0.21) 11.66ms
2018-04-23 21:30:46,300 - WARNING  404 GET /apple-touch-icon-152x152.png (192.168.0.21) 22.09ms
2018-04-23 21:30:46,325 - WARNING  404 GET /apple-touch-icon-precomposed.png (192.168.0.21) 12.13ms
2018-04-23 21:30:46,348 - WARNING  404 GET /apple-touch-icon.png (192.168.0.21) 11.73ms
2018-04-23 21:30:46,369 - WARNING  404 GET /favicon.ico (192.168.0.21) 11.53ms
2018-04-23 21:51:53,961 - WARNING  404 GET /favicon.ico (192.168.0.46) 40.27ms
2018-04-23 21:51:55,421 - WARNING  404 GET /favicon.ico (192.168.0.46) 11.67ms
2018-04-23 21:51:59,114 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-04-23 21:59:30,267 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
`

heres my musicbox config
`################################# 
# Pi MusicBox / Mopidy Settings # 
################################# 
# 
# Edit the settings of your MusicBox and the Mopidy music server here
# This is a file read by both the MusicBox startup-scripts and Mopidy.
# For more info on the settings of Mopidy: http://docs.mopidy.com/en/latest/config/
# or the particular extenion's GitHub page 
#
# Every line starting with a # is a comment, it does not do anything until you remove the #

# --------------------
# | Network Settings |
# --------------------
[network]
# Settings for your WiFi network, if you use a (supported) wifi-dongle
# Only supports WPA security, no WEP or access points without security (dive into the command line for that!)
wifi_network = virginmedia4550587
wifi_password = mqs6mxfwXkXy
# Optionally set the wifi region for correct regulatory configuration (available channels etc.)
# Use the ISO / IEC 3166 alpha2 country code, e.g. wifi_country = GB
wifi_country = GB

# Set the name of the MusicBox. 
# In this way you can identify and access different devices on the same network e.g. across different rooms.
# A MusicBox device named kitchen would be accessible from a web browser at http://kitchen.local/, from an MPD 
# client at kitchen.local and advertised as kitchen on AirTunes.
# The name is restricted to a maximum of 9 alphanumeric characters (no spaces, dots, etc).
# You can even have different devices with different Spotify accounts when needed.
name = MusicBox

# Mount Windows Network Drive on boot (via samba/cifs)
# The address is exactly how Samba mount works: //servername/mountpoint/directory
# e.g. //192.168.1.5/musicmount or //server.local/shared/music
# if the mount needs a username/password, also set it (leave empty for guest-access)
# Don't forget to let Mopidy/MusicBox scan the contents at first boot (see below) 
mount_address = \\media-server\usb
mount_user = pi
mount_password = daniel19

# Here you can change the default workgroup of the Windows network.
# This will set the workgroup to the name you want
workgroup = WORKGROUP

# Enable this to allow remote login via SSH on MusicBox
[... 257 lines cut ...]
config_file = /etc/mopidy/logging.conf
debug_file = /var/log/mopidy/mopidy-debug.log

[local]
enabled = true
library = sqlite
media_dir = /music
scan_follow_symlinks = true
excluded_file_extensions = .directory .html, .jpeg, .jpg, .log, .nfo, .png, .txt, .mkv, .avi, .divx, .qt, .htm, .zip, .rar, .gz, .pdf, .exe, .ini, .mid, .db, .m3u, .sfv, .midi

# WARNING: Do not add settings for [local-sqlite]
# because it breaks the startup script (won't read dashes in section names)

[file]
media_dirs = /music|Music
follow_symlinks = true

[m3u]
playlists_dir = /music/playlists
[subsonic]
enabled = false
ssl = no
`

I'm using musicbox to stream TuneIn through icecast2