pimusicbox/pimusicbox #473: Release for Pi 3 B+

Letterkerl, 2018-03-19:
As I saw, support for the new pi was already added. However, the latest release is from July last year, making it impossible to directly install MusicBox to a Pi 3 B+.

kingosticks, 2018-03-19:
Yes, I am [planning on releasing an update tonight](https://discourse.mopidy.com/t/spotify-connect-error/2314/4).

kingosticks, 2018-03-20:
I've released a new version at https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC6

Letterkerl, 2018-03-20:
Great, thank you. I'll test it later.

Letterkerl, 2018-03-20:
Seems to work for the most part. However I don't manage to mount my NAS.

`Mounting Windows Network drive: //nas.local/Public/Shared Music/ ...
mount error(22): Invalid argument
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
`
I tried to manually set the adress in the startup.sh and also double checked with my working Pi 1 B.

kingosticks, 2018-03-20:
It's always the thing you don't test! 

Seems that changing `sec=ntlm` to `sec=ntlmssp` (or removing it entirely) seems to do the trick for me with my Ubuntu SMB share. What OS are you using on your file server?

kingosticks, 2018-03-20:
Looks like the better fix is to remove the sec argument entirely since it was wrong. See https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1746482

kingosticks, 2018-03-20:
Fixed in https://github.com/pimusicbox/pimusicbox/commit/832a56a997c8e05f2e2e4fc603f4d4828d981a38
```
wget -O /opt/musicbox/startup.sh https://raw.githubusercontent.com/pimusicbox/pimusicbox/develop/filechanges/opt/musicbox/startup.sh
```

Letterkerl, 2018-03-21:
Yes, looks good now. Thank you!

closed 2018-03-21 by Letterkerl

DonLobster, 2018-03-22:
I still can't mount a network drive manually, even after updating the startup.sh and adding "mount options = """ in settings.ini and omitting sec argument :/

My mount line looks as follows:
mount -t cifs -o username=user,password=pass //192.168.1.140/download /home/gmax/DISKSTATION/download

I get the following error message:
mount error(112): Host is down
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

Any suggestions?
TIA!

kingosticks, 2018-03-22:
Can you try adding the extra option `vers=2.0`

kingosticks, 2018-03-22:
Or 1.0 actually

DonLobster, 2018-03-22:
YES! Version 2 didn't work, but vers=1.0 did :-D Thanks again for the quick reply!

kingosticks, 2018-05-23:
Assuming you also patched startup.sh (as per https://github.com/pimusicbox/pimusicbox/issues/473#issuecomment-374779714) then it should have worked like it has for everyone else. I'm about ready for RC7 now anyway so this will (finally) go away.

commit 7979b7b9 (2018-03-22): Default mount option which should work for most people. (Fixes #473).