pimusicbox/pimusicbox #474: Boot loop on Raspberry Pi 3 B

aidanb, 2018-03-22:
I am trying to run version 7RC6, however it will not boot up and keeps restarting indefinitely.

I have tested the hardware and SD card and everything works fine with the latest Raspbian image.

I checked /var/log/syslog on the SD card but all the directories are empty.

Any thoughts on how I should proceed?

kingosticks, 2018-03-22:
Ideally plug in a monitor. Failing that, is there a /var/log/musicbox_startup.log file?

aidanb, 2018-03-22:
Wow, thanks for the quick reply.

/var/log/musicbox_startup.log does not exist.

I watched it boot up with a monitor attached and the only error I saw was "fsck died with exist status 1" followed by "failed (code 1)" in red.

aidanb, 2018-03-22:
I changed the power supply and it works now. Weird.

closed 2018-03-22 by aidanb

kingosticks, 2018-03-22:
It is odd since it's happening so very early in the boot process, I can't really think what we do differently at that point. Maybe some of the config.txt options makes the difference. But anyway, glad it's working. Thanks for updating this thread.