pimusicbox/pimusicbox #479: YouTube no search

SanKen, 2018-05-23:
I have a problem.... PI music BOX no search He does not search for any songs. The only thing that displays it.

"Loading Data From (IP) Please wait" And nothing more.... with default settings. let someone tell me what to enter into the browser. http://ip/musicbox_webclient/index.html?select-add=1#search

SanKen, 2018-05-23:
How to fix this? Only upgrade?

kingosticks, 2018-05-23:
I'm afraid there's currently no fix. Upgrading doesn't help, this is the every latest version of the extension. Youtube decided to revoke the mopidy-youtube API key, see https://github.com/mopidy/mopidy-youtube/issues/92

SanKen, 2018-05-23:
what is
"I've been working on migrating from pafy to youtube-dl, you can see my progress at https://github.com/seth--/mopidy-youtube/tree/youtube-dl"

kingosticks, 2018-05-23:
What do you mean? It's exactly what it says it is.

SanKen, 2018-05-23:
I wonder if it works together with mopidy. Is there any other way around to play music with yt.

kingosticks, 2018-05-23:
You can still *Browse* or you can discover single videos elsewhere (e.g. through your web browser) and enter the http link with the string "yt:" prepended, as described in the [README](https://github.com/mopidy/mopidy-youtube/blob/develop/README.rst#usage), on the webclient *Streams* page. You just can't search (or lookup youtube playlists?)

If you want to try out that experimental youtube-dl version you can install it on your pimusicbox system by running 
```
pip install --no-deps --upgrade https://github.com/seth--/mopidy-youtube/archive/youtube-dl.zip
```
I have no idea what the state of that work is and I wouldn't suggest trying this unless you know what you are doing... and please direct any specific problems/questions regarding it, directly to the author.

SanKen, 2018-05-23:
That's enough for me. But this add link not working :(

he recognizes the song, but time does not go. Only (sync)

kingosticks, 2018-05-23:
The track loading but failing to actually play is usually due to being an incompatible audio format. There's a particular format that doesn't work in pimusicbox's version of gstreamer and that seems to be more and more common. It's just luck (or lack of) if that's the one that's being used for your video. I don't have a fix for that yet.

SanKen, 2018-05-24:
please send me link which works for you.

nglgzz, 2018-09-13:
Are there plans to fix this? What is the actual problem? I would love to use Youtube search on the web client.

natumbri, 2019-06-24:
@nglgzz @SanKen Any interest in trying the new fork I made?  More details over at mopidy/mopidy-youtube#103

I would recommend the [feature/proxy](/natumbri/mopidy-youtube/tree/feature/proxy) branch at this stage - it respects mopidy's proxy configuration.

nglgzz, 2019-07-30:
Hey @natumbri, how can I try your fork? I'm gonna reinstall pimusicbox soon.

natumbri, 2019-07-30:
At this stage, I'd suggest installing the official master version, and then
copying the files from my fork over the top. Should work...